hawku/TabletDriver #1: Driver Stopped, Restarting!

albixow, 2018-03-12:
Did message you about this on reddit but you didn't reply eventually so thought I'd post it here too. #

https://albi.s-ul.eu/1DPT4rO7

Tried everything I can think of, all my Wacom drivers are uninstalled, tried restarting pc, running as admin, nothing works. Just getting the same error.

hawku, 2018-03-13:
Uninstall the "Wacom Tablet" device from the Window Device Manager.
Check the "Delete the driver software for this device." checkbox when uninstalling the device.

albixow, 2018-03-13:
Tried it, then restarted PC, installed your driver, opened the Driver GUI and still the same error.

Calideon, 2018-03-13:
I have same issue, but my tablet is connected. There is screeenshot of my problem: https://puu.sh/zGNnX/093324cd3b.png

brimstone-, 2018-03-13:
With TabletDriver not yet installed, I uninstalled my current driver for my H420 with that particular checkbox ticked, Windows asked me to reboot, I unplugged my tablet before reboot, installed both the main driver and the huion one, rebooted, plugged in my tablet, launched the GUI, and am still getting the same error above. I also tried the tablet.cfg change in the reddit thread.

derekxwu, 2018-03-13:
Reinstalling and re-uninstalling the Wacom driver (restarting both times) helped me with this problem.

Meigs2, 2018-03-13:
Yeah, I'm getting the same issue here. Cant fix it. Let me see if updating windows helps and I'll report back.

Meigs2, 2018-03-14:
WAIT Updating to the fall creators update fully fixed my issues. Try FULLY updating windows for the people having this issue!

hawku, 2018-03-14:
@Meigs2 That is interesting. That might be the reason for these detection problems, because only few people seems to have the problem after uninstalling the drivers properly.
Are you using the latest version of the driver (0.0.13 or 0.0.14)?

Version 0.0.13 adds new tablet configurations (config\wacom.cfg), so that might have improved the Wacom tablet detection.

Please, right click the console and press Copy all and paste the "Tablet open!" part here. It should look like this:
```
2018-03-15 00:26:02 [INFO] >> #
2018-03-15 00:26:02 [INFO] >> Tablet 0x056a 0x037a 0xFF0D 0x0001
2018-03-15 00:26:02 [INFO] Tablet open!
2018-03-15 00:26:02 [INFO] >> Name "Wacom CTL-472"
```
Copy all function doesn't work in the 0.0.13, so first you need to update the driver to version 0.0.14.

Meigs2, 2018-03-15:
Nope, was using 0.0.12

dylanjerry, 2018-03-26:
I uninstalled all Tablet drivers and I have installed the vmulti device and reinstalled it multiple times
here is my problem 

[image]

Gerodar, 2018-04-03:
None of the solutions here helped me. Cannot find any traces of old drivers on my pc (uninstalled and installed vmulti multiple times) still doesnt help. 
Using CTL-471

nutorate, 2018-05-22:
> Uninstall the "Wacom Tablet" device from the Window Device Manager.
Check the "Delete the driver software for this device." checkbox when uninstalling the device.

I can't find wacom tablet in my device manager. In which folder is it?

melloeOsu, 2018-06-16:
Hey, maybe I'm just an idiot and this is a stupid quetion but..
Where is config\tablet.cfg? I can't find it to make the proper edits to it
I use a CTL-460, and I downloaded the driver from the wacom website

BateauSD, 2018-08-14:
[image]
Hmm

woahsia, 2018-10-22:
2018-10-23 00:42:23 [ERROR] Can't open VMulti device!

2018-10-23 00:42:23 [ERROR] Possible fixes:
2018-10-23 00:42:23 [ERROR] 1) Install VMulti driver
2018-10-23 00:42:23 [ERROR] 2) Kill PentabletService.exe (XP Pen driver)
2018-10-23 00:42:23 [ERROR] 3) Uninstall other tablet drivers and reinstall VMulti driver
Driver stopped. Restarting!
2018-10-23 00:42:29 [ERROR] Can't open VMulti device!

2018-10-23 00:42:29 [ERROR] Possible fixes:
2018-10-23 00:42:29 [ERROR] 1) Install VMulti driver
2018-10-23 00:42:29 [ERROR] 2) Kill PentabletService.exe (XP Pen driver)
2018-10-23 00:42:29 [ERROR] 3) Uninstall other tablet drivers and reinstall VMulti driver
2018-10-23 00:42:30 [ERROR] Can't open VMulti device!

2018-10-23 00:42:30 [ERROR] Possible fixes:
2018-10-23 00:42:30 [ERROR] 1) Install VMulti driver
2018-10-23 00:42:30 [ERROR] 2) Kill PentabletService.exe (XP Pen driver)
2018-10-23 00:42:30 [ERROR] 3) Uninstall other tablet drivers and reinstall VMulti driver
Driver stopped. Restarting!

I was using an xp pen but it spoilt so I switched to my wacom cth 480. PRior to this I used the xp pen drivers, but decided to switch to these drivers for the wacom tablet. I uninstalled all my other drivers before installing this but I'm still getting these errors, nothing on this thread has helped so far

ARToftheUSA, 2018-11-09:
Was anyone able to solve the Kill PentabletServices.exe Issue?????

Fogazzle, 2020-06-18:
I am using a xp pen g640 and i am having the same problem and nothing has fixed it.

JustinMiroux, 2020-06-24:
I am using a star  g640 and have also the same problem can you help ?

rasters10, 2020-08-20:
> I am using a star g640 and have also the same problem can you help ?
This guy found a solution to the problem:
https://github.com/hawku/TabletDriver/issues/602#issuecomment-607175581

rasters10, 2020-08-20:
> I am using a xp pen g640 and i am having the same problem and nothing has fixed it.

> > I am using a star g640 and have also the same problem can you help ?
> > This guy found a solution to the problem:
> > [#602 (comment)](https://github.com/hawku/TabletDriver/issues/602#issuecomment-607175581)

rasters10, 2020-08-20:
After this my g640 start working at hawku driver

MuNgUmAn, 2020-10-20:
> I uninstalled all Tablet drivers and I have installed the vmulti device and reinstalled it multiple times
> here is my problem
> 
> [image]

Happens to me too, and i have a Wacom CTL - 472 and I have deleted all my old drivers and it still doesn't work

awevxz, 2020-11-11:
Hey, idk if this will work for anybody else but it works for me:)

If you turn on run at windows startup in the GUI and keep your tablet physically connected, shut down your pc and then turn it back on, don't log in until it's completely booted up, I wait like at least 2 minutes just to make sure it has booted fully (I have a shit pc btw). The driver should connect to your tablet automatically and start working.  I don't know if this will work for anybody else but just make sure you've installed vmulti_drivers and that your tablet is supported. 

I'm using the Wacom CTL-4100 and running windows 10.

alexdis01, 2020-11-27:
I've just bought an Xp-pen Star G430S and it gives me this issue, how can I resolve?
[image]

landro-trying, 2020-12-25:
> I've just bought an Xp-pen Star G430S and it gives me this issue, how can I resolve?
> [image]

hi I have the same problem and same tablet do you find any solution?

alexdis01, 2020-12-25:
> > I've just bought an Xp-pen Star G430S and it gives me this issue, how can I resolve?
> > [image]
> 
> hi I have the same problem and same tablet do you find any solution?

Actually i didn't find a solution, but i've started to use the official driver and it works without problems so i advice you to do the same

inrate, 2020-12-26:
I have a problem with my xp-pen G640 as it seems that it is working off some type of windows driver that I can't really find and delete. 
[image]
Does anyone know how to solve this problem?

SwagBrad, 2021-01-14:
Umm, so what files do we delete to remove Wacom drivers.

simptsuki, 2021-03-10:
https://www.wacom.com/en-us/support?guideTitle=How-do-I-uninstall-(manually)-and-re-install-the-Wacom-driver-on-Windows-for-a-Pen-Tablet%2C-Pen-Display%2C-or-Pen-Computer%3F&guideId=010-373 here. also in the tutorial it says that you have to empty the recycling bin. i had to do that in order for it to work

ssebastianoo, 2021-03-22:
I have similiar problems (Wacom **CTL-4100**), when I delete the **Wacom Drivers** from the device manager, **TabletDriver** will stop working too.

I uninstalled it from the device manager
[image]

_With_ **Wacom Drivers** installed (it doesn't work)
[image]

_WIthout_ **Wacom Drivers** installed
[image]

If I un-plug re-plug the tablet, wacom drivers will reinstall automatically.

PixelHir, 2021-09-27:
I went into the driver_vmulti folder, opened a command prompt with Administrator privileges, and ran a command `devcon.exe /r install vmulti.inf pentablet\hid` and it worked.
It seems that the installer failed for some reason.