hawku/TabletDriver #10: The GUI crashing on startup

dedddd, 2018-03-13:
So, I've uninstalled my Wacom drivers. I've tried many things to see if it's fixed, like reinstalling and restarting my PC to see if it works, but nothing does. Idk if it matters but I'm using w7 and CTL-471 https://puu.sh/zGPHA/b088903394.png

Shoko84, 2018-03-13:
I've heard that there are some problems about _drivers_ on W7. It may have some issues when loading drivers and then instant crash 🤔 

Maybe @hawku does have an idea about why ?

dedddd, 2018-03-13:
Haven't tried 12, I'll have a look

hawku, 2018-03-13:
@dedddd Upload the startuplog.txt from the drivers's folder to here.

dedddd, 2018-03-13:
Since 12 seems to not work for me, I'm trying 11. Here's the text file https://puu.sh/zGU3U/c356a03a96.txt

hawku, 2018-03-14:
@dedddd Please, try the latest version: http://hwk.fi/TabletDriver/TabletDriverV0.0.14.zip

dedddd, 2018-03-15:
Well, it still seems to crash but here's the startuplog https://puu.sh/zI7Vr/33fd482c8d.txt

hawku, 2018-03-17:
@dedddd Try the latest version. I've added more debug information to the startuplog.txt

dedddd, 2018-03-22:
I dunno, it's still crashing. Guess I'll just stick to the Wacom drivers

Banbeucmas, 2018-03-22:
@dedddd startuplog.txt is for debugging. So I think you should reopen the driver and privide the new startuplog

closed 2018-04-09 by hawku