hawku/TabletDriver #12: Tablet not found!

Steinosu, 2018-03-13:
In the console it says: "Can't open HID tablet [usb ids]" and i really don't know what to do :(

Shoko84, 2018-03-13:
Which tablet ? OS ? Devs needs some informations about why it's not working etc..

Steinosu, 2018-03-13:
https://postimg.org/image/mult9hbtx/

Shoko84, 2018-03-13:
Just copying pasta from another issue :

> Add these lines to the Commands tab's first field and press Apply:
> 
> - log debug.txt
> - debug 1
> 
> That should create a debug.txt log file which contains lots of tablet status information.
> Try to reproduce the issue when the debug log is running.
> Log file will be huge if you play too long with logging enabled, but the log file will be cleared every time you apply or save settings.
> When you have captured the issue, close the GUI without saving or applying, zip the log file and upload it here.

This might help @hawku, but I also saw that you have a CTH-680, which it couldn't be tested properly iirc, I didn't get much into his code so I don't know why it could not work.

Are you sure you've uninstalled your Wacom drivers before installing TabletDriver's drivers ? (the multi one)

Steinosu, 2018-03-14:
But nothing happens when I'm typing this in (Win 10; uninstalled wacom drivers)

Shoko84, 2018-03-14:
Like I copy pasta'd, you should write these two lines and then click on the 'Apply' button at the buttom of the window.

It should create a debug.txt file, zip it and upload it here :wink:

hawku, 2018-03-14:
@Steinosu Which Wacom tablet do you have?

closed 2018-03-16 by hawku