hawku/TabletDriver #15: Driver Crashing when pressing 'SAVE'

Amblis, 2018-03-13:
I uninstalled my Wacom Drivers.
Restarted my computer.
Installed vmulti drivers.
Launched drivers, after setting up my settings it crashed when pressing 'SAVE'
-went offline
second day second try,
pressed 'SAVE', crash again.

Shoko84, 2018-03-13:
Just copying pasta from another issue :

> Add these lines to the Commands tab's first field and press Apply:
> 
> - log debug.txt
> - debug 1
> 
> That should create a debug.txt log file which contains lots of tablet status information.
> Try to reproduce the issue when the debug log is running.
> Log file will be huge if you play too long with logging enabled, but the log file will be cleared every time you apply or save settings.
> When you have captured the issue, close the GUI without saving or applying, zip the log file and upload it here.

Also, you should also write down on which OS you're running and which version of the TabletDriver you have (if it's not the latest one), this might help !

Amblis, 2018-03-14:
I'll try this out when I get home again today.
Currently on latest version of Windows 10 Pro and 0.0.12 on the tablet drivers. I'm using a CTH 480

Amblis, 2018-03-14:
where exactly can I see the .txt file you said it creates? seen as nothing is wrong with the program as long as I don't press 'SAVE' in the bottom left corner. my problem is the fact that I have to redo my settings every single time since I can't save without the program crashing and I saw some things happening in the console in the program itself but since it crashes when I press save, I won't be able to see what comes out

hawku, 2018-03-14:
I added error handling to the configuration saving in the latest version:
http://hwk.fi/TabletDriver/TabletDriverV0.0.13.zip

You might not have write permissions in the TableDriver's directory.

closed 2018-03-18 by hawku