hawku/TabletDriver #3: Is CTE-450 support possible?

acm321, 2018-03-13:
_(I also posted this as a reply on Reddit but I figured it wouldn't hurt to post here as well)_

The driver recognizes the tablet correctly and provides basic functionality using this config (I had to guess a few values so that may be why it doesn't work perfectly):

`#`
`# Wacom CTE-450`
`#`
`Tablet 0x056a 0x0017 0x000D 0x01`
`Name "Wacom CTE-450"`
`ReportLength 10`
`ButtonMask 0xE0`
`MaxX 14760`
`MaxY 9230`
`MaxPressure 511`
`Width 147.6`
`Height 92.3`

The settings in the GUI don't do anything (not that I expected it to work at all since it's not officially supported), and it seems to be stuck in relative positioning.

Other than that it works fine and recognizes clicks nicely, about 10ms faster than the official wacom driver. It's clear that the Wacom driver is using quite a bit of smoothing which this was able to eliminate.

Here is the console output on startup: https://i.imgur.com/zcL6GcM.png

TL;DR: CTE-450 works but only in relative positioning, do I just need to tweak the config or would you have to code it in to support it?

MineRollZ, 2018-03-15:
Is there anyway you could convert this to the CTE-650? In my logic I'd expect it to be pretty easy since it's a larger version of the 450. How did you find these numbers to plug into the config?

hawku, 2018-03-15:
@acm321 Does the tablet work with different ButtonMask setting?
Can you try this:
`ButtonMask 0x40`

I'm going to add the tablet to the automatically generated config (wacom.cfg), but i don't know if the tablet will work with that ButtonMask setting.

acm321, 2018-03-17:
@hawku No, 0x10 is the only value I've gotten to work with that setting. I wasn't sure where to find the "correct" value so I had tried several and settled on 0x10. Others produce incorrect (bad mapping) or non-functional behavior.

hawku, 2018-03-17:
Have you tested the latest driver version?

This is the current automatically generated configuration (config\wacom.cfg) for the CTE-450 :
```
# Wacom CTE-450
Tablet 0x056A 0x0017 0xFF0D 0x0001
Tablet 0x056A 0x0017 0x000D 0x0001
Name "Wacom CTE-450"
ReportLength 9
ButtonMask 0x00
MaxX 14760
MaxY 9225
MaxPressure 511
Width 147.600
Height 92.250
InitFeature 0x02 0x02
```

Wacom linux driver code:
https://github.com/linuxwacom/input-wacom/blob/master/3.7/wacom_wac.c#L381

It looks like that the `ButtonMask 0x80` might work.

acm321, 2018-03-18:
@hawku ButtonMask 0x80 indeed seems to work on version 0.0.18.

closed 2018-03-26 by hawku