hawku/TabletDriver #4: Screen mapping doesn't work properly

kevinwlau, 2018-03-13:
I have two monitors at 1080p. 

- Using the `Full desktop` option only maps the full tablet area to monitor 1
- Using `DISPLAY 1` maps the full tablet area to the left half of monitor 1
- Using `DISPLAY 2` maps the full tablet area to the right half of monitor 1

[image]

hawku, 2018-03-13:
Which version of Windows are you using?

I had that problem with Windows 7 and made a fix for that, but haven't yet figured out why that also happens on other Windows versions.

Workaround for that problem is to edit the Config.xml and set the DeveloperMode to true.
`<DeveloperMode>true</DeveloperMode>`

After that you should have desktop size settings at the bottom of the GUI:
http://hwk.fi/TabletDriver/DesktopSize.png

Uncheck the "Auto" checkbox and edit values to match with the screen area the driver can reach.
Set the screen map size to those same values.
I think that the correct values for you should be Width=1920  Height=1080

kevinwlau, 2018-03-13:
I'm using Windows 8.1

The `DeveloperMode` workaround works partially- the tablet now maps the area correctly across the two monitors. However, the 2nd monitor is still inaccessible with the tablet - the cursor gets stuck at the right edge of monitor 1.

[image]

Banbeucmas, 2018-03-15:
Tbh, I would prefer a mapping solution which provide hot key to switch mapping with each screen.

hawku, 2018-03-15:
@kevinwlau You also have to set the screen map width to 1920.

I'm going to make a cursor position validator that checks if the vmulti can use the full desktop area.
That might need lots of testing with different Windows versions, so it isn't a quick fix.

kevinwlau, 2018-03-15:
@hawku I want to clarify that the settings I have screenshotted with developer mode works just as I intended. Except that I can't do anything with the second monitor.

hawku, 2018-03-17:
@kevinwlau Windows USB/HID absolute mouse handling acts differently on different systems.
On Windows 7 the absolute mouse position maps only to main monitor, but on later Windows versions it should map to the whole virtual desktop space.
Haven't tested the driver on Windows 8 or 8.1, so it be that the 8 and/or 8.1 might have the same problem as Windows 7.
There also might be a 3rd party mouse driver that replaces the default Windows absolute mouse behavior.

I have now enabled the desktop size settings to everyone in the version 0.0.16.

kevinwlau, 2018-06-22:
I've upgraded to Windows 10 and this issue no longer occurs.

Still valid for Win 8.1 though

Qylwok, 2018-08-16:
I didn't have this kind of issue when I was using Windows 10 Family on my HDD, but now that I use Windows 10 Education on my SSD instead I have this same issue.

I have 2 screens : 1920x1080 and 1680x1050, and the 1680 one now can't be reached by my tablet for some reasons, even when I modify the Desktop Size Width&Height and/or the Screen Map Width/Height to 1920x1080 or 1680x1050 with DeveloperMode true.

I am using a CTH-470 and the 0.1.5.4 driver version on Windows 10 Education.

Here is a screen of my TabletDriverGUI : https://i.imgur.com/P3Qa5Xw.png

hawku, 2019-01-09:
Latest driver versions does have SendInput/Compatibility mode that might fix this problem.

closed 2019-01-09 by hawku

Vincent-wuu, 2021-06-10:
[image]
my driver all of a sudden has stopped working properly. When i reach the end of my tablet, it only reaches halfway on my monitor.

I am running on Windows 10

Sortixx, 2021-06-10:
I have the exact same issue. Did you, found the Problem or a Solution  ?

spoctral, 2021-06-10:
> [image]
> my driver all of a sudden has stopped working properly. When i reach the end of my tablet, it only reaches halfway on my monitor.
> 
> I am running on Windows 10

i am having the same exact issue aswell, i have no solution.

rzoudrip, 2021-06-10:
Same problem, did u guys update windows? theres an windows update since today and after i did it, the drivers only worked half a monitor. im very sad i cant play osu :CC

tsengsys, 2021-06-10:
I am having the same problem as well. Didn't have this problem a few hours before the windows update. Still couldn't tell what exactly is causing the issue.

JoeyXD95, 2021-06-10:
Same issue aswell. Just updated my Windows.

Elf404, 2021-06-10:
Yep me too, updated windows last night. The screen maps both my monitors on my main one. Quick fix for me was to use full desktop but hoping for an actual fix as it just feels off. :/

H4ED, 2021-06-10:
Hi, having the same problem since the windows update

CliffordTBRC, 2021-06-11:
same issue here

CliffordTBRC, 2021-06-11:
just swapped to OTD, it works fine seems like the windows update has broken it. https://github.com/hawku/TabletDriver/issues/1087

H4ED, 2021-06-11:
Any solutions ? I need to close one of my monitor if I want to use my tablet

Yeetedscrub, 2021-06-12:
try downloading .NET runtime from [https://dotnet.microsoft.com/download/dotnet/5.0/runtime/?utm_source=getdotnetcore&utm_medium=referral](url)
you need to choose the one that says "run desktop apps"

drago20013, 2021-06-12:
What worked for me is setting the Desktop size manually to 1920x1080 and in the screen map Set Area to Full desktop.

pyjthon, 2021-06-12:
Tried downloading the .net didnt work and setting the desktop area didnt work too

Hedgy134117, 2021-06-12:
> What worked for me is setting the Desktop size manually to 1920x1080 and in the screen map Set Area to Full desktop.

This fixed it for me. Thanks!

Eveaon, 2021-06-14:
Hey everyone, I was suffering with the same problem, what worked for me was downgrading windows. Here is the tutorial I used https://www.techadvisor.com/how-to/windows/downgrade-windows-10-3696498/ Good luck osu! gamers

Filles01, 2021-06-14:
Here's my fix (I have two monitors, CTL-672):
https://youtu.be/NWMCEBjCSWs

Win+P select only one screen, in Hawku's tablet drivers select "DISPLAY1" save, and then you Win+P again a select "Extend"

sprain3d, 2021-06-17:
I'm having the same problem but my smoothing feels like its 125hz even though its put at 1000hz

moshier1, 2021-06-19:
windows update broke it for me too, if I turn off auto for the desktop size and put in my main monitor manually it works, but it still feels off(less sensitive, mostly just in the y direction)

sheeezz, 2022-06-13:
[best projector for golf simulator](https://tekclue.com/best-short-throw-projector-for-golf-simulator/
**good work very helpful**

sheeezz, 2022-06-13:
**<a herf="[url](https://tekclue.com/best-short-throw-projector-for-golf-simulator/)" >best projector for golf simulator</a>**

sheeezz, 2022-06-13:
best projector for golf simulator([url](https://tekclue.com/best-short-throw-projector-for-golf-simulator/)