hawku/TabletDriver #5: Can run multiple instances of the driver

Zinkton, 2018-03-13:
If you launch a 2nd instance and change the area on it, both area from instance 1 and area from instance 2 will be active and you'll have 2 cursors. May be fixed by just maximizing the current instance when trying launch the program?

Shoko84, 2018-03-13:
Requested a commit about that issue, can be merged into the master branch if it hasn't already been done locally from @hawku 's side 👍

hawku, 2018-03-13:
@Shoko84 thanks for the pull request!
I had a different solution for the problem, but I really appreciate any pull request for this project!

closed 2018-03-13 by hawku

Shoko84, 2018-03-13:
I would be glad to help you and to contact you anytime if you have some few spare issues (well idk where we could be anyways :D)