hawku/TabletDriver #9: Gaomon S56K receiving random cursor lags

MesAyaneru, 2018-03-13:
The lag pops out sporadically, though it affects my gameplay pretty much, like making me choke.

insomnyawolf, 2018-03-13:
Try searching for a pattern like where and how the lag happen.
Actually dev cant do anything with such few information.

hawku, 2018-03-13:
@MesAyaneru i can't reproduce the issue with my Gaomon S56K.

Add these lines to the Commands tab's first field and press Apply:
* log debug.txt
* debug 1

That should create a debug.txt log file which contains lots of tablet status information.
Try to reproduce the issue when the debug log is running.
Log file will be huge if you play too long with logging enabled, but the log file will be cleared every time you apply or save settings.
When you have captured the issue, close the GUI without saving or applying, zip the log file and upload it here.

MesAyaneru, 2018-03-14:
alright thanks for your instruction, I've also enclosed my replay and a
screen capture of the point where the lag occurred. You should be able to
see I sliderbroke abruptly.

hawku, 2018-03-14:
@MesAyaneru I don't see the replay or screen capture. Did you upload those here?

MesAyaneru, 2018-03-15:
I did put them in the zip document

hawku, 2018-03-18:
@MesAyaneru Attach the zip file to GitHub issue comment. Do not try to send it through email

closed 2018-03-26 by hawku