hawku/TabletDriver #18: CTL-470, Cursor Shaking

GDJake, 2018-03-14:
The Cursor keeps shaking. Barely noticable while using Full Area, but it gets more intense on small Inputarea. "Raw Input" is on, "Map absolute raw input to the osu! window" is turned off.
Not that huge of a Problem, but might cause Issues on high CS.
Otherwise the Driver works really well, way better than Wacom 470 software.

SuperTilted, 2018-03-14:
https://www.reddit.com/r/osugame/comments/83hbiq/testing_different_tablet_drivers_with_a_3d_printer/dvjj78s/

According to Hawku, that should just be the normal signal "noise" that comes with interaction between the tablet and pen. Since his driver doesn't have any smoothing to begin with, the noise becomes more apparent than compared to a Wacom driver. You can try increasing the latency on the smoothing filter to help mitigate that shaking but it doesn't get rid of it completely, based on my experience.

hawku, 2018-03-14:
@SuperTilted Thanks for reply!
I'm going to add the frequently asked questions to the readme.

closed 2018-03-14 by hawku