hawku/TabletDriver #23: ctl-490 hover distance & tablet click

Yentis, 2018-03-14:
With these drivers for some reason my hover distance is far lower than with wacom drivers.
I'd say it's about 5cm with wacom while with these it's only like 2cm.

Also when I accidentally tap the tablet my cursor messes up a bit, I've seen a lot of reports about this being the case even on the wacom drivers but I only have this issue here.

I've attached my wacom config, area is the exact same and all buttons that can be disabled are disabled.
[Backup.txt](https://github.com/hawku/TabletDriver/files/1812824/Backup.txt)

Shoko84, 2018-03-14:
For the first point, I got the same "feeling" of having a smaller area. To be honest, for 3 days, I was messing up everything, even having the same area. Today, I decided to make it bigger from 125.7mm to 133mm width size speaking.

The reason is because this driver, compared to the Wacom drivers, are removing all the _smoothing_ filters applied on the cursor, which make it more stable, less shaky, but with more latency.
With this one, nothing is applied on the cursor but makes it _raw_ if you know what I mean.

For the last one tbh I don't know, you should wait an answer from @hawku or from somebody who's having the same issue, I do have a CTL480 so I can't help you much about that 😕 
But I'm also seeing my cursor shaking more than the Wacom's drivers when hovering _(more noticeable when hovering at about 3 to 4mm from the tablet)_, but that's a normal thing and shouldn't be worried about (except if it's really **that** much)

hawku, 2018-03-15:
@Yentis Are you sure that the maximum hover height is 5 cm with Wacom drivers?

I do have about 1 cm (10 mm) maximum hover height on my CTL-490 with both drivers.

I do have that tap problem with my CTL-490 too, but i think it's a hardware problem.

Best would be just to tape the pen tip, so its always pushed down. That also disables the hardware smoothing done by the tablet itself.

Yentis, 2018-03-15:
Seems like the hover distance is about the same now since updating to the latest version.

I never had to tape down the tip with the wacom drivers, but that might have just been because I had more hover distance to work with.

Peachy-Cai, 2018-03-16:
im fairly certain the hardware input lag varies across the different variants of the 490, some of my friends have a 490 but a different variant code and they dont tape the nib yet hover without noticeable (to them) lag

closed 2018-03-18 by hawku