hawku/TabletDriver #24: XP PEN G430 Tablet not found

pytajnik2, 2018-03-14:
^help please

pytajnik2, 2018-03-14:
i have same issue with new xp pen driver "XP-PenWin(20180130)", older driver works

hawku, 2018-03-15:
@pytajnik2 Try to run the TabletDriverGUI as an administrator.

Are you sure that your tablet is G430 and not the newer model G430**S**?

Can you find tablet's HID device from the Windows Device Manager?
http://hwk.fi/TabletDriver/WacomHID.png, but the device instance path should start with `HID\VID_28BD`

pytajnik2, 2018-03-15:
its g430 not g430s and i trying run as an admin
i can find talbet in usb section not HID section https://i.imgur.com/z8YWqSd.png
in the HID section, I have something like that https://i.imgur.com/p9PB1rx.png

hawku, 2018-03-17:
@pytajnik2 There are at least two different XP-Pen G430 models. You have the older model that is not supported by this driver or XP-Pen's latest drivers. Check the issue #35

hawku, 2018-03-17:
Duplicate of #35

closed 2018-03-17 by hawku