hawku/TabletDriver #27: Pen's Report Rate for the CTH-480

SuperTilted, 2018-03-14:
I'm not sure if this should be considered an issue or if it was intended but I feel like it should be addressed anyways.
 
The tablet's original report rate is somewhere around ~133pps. According to the pattern tests you've done on the other Wacom drivers, this matches that description. I checked my tablet's report rate while on TabletDriverV0.0.13 and it's running at about ~500pps. I've also tested a Wacom driver (6.3.15-1) after and it matches the tablet's original report rate of ~133pps. I've uploaded the results of both tablet drivers to see. I used a mouse rate checker (http://www.softpedia.com/get/System/System-Miscellaneous/Mouse-Rate-Checker.shtml) as well as osu!'s ingame report rate checker in the screenshots. 

The higher report rate actually sounds great but I'm worried about whether or not this could affect the tablet's lifespan. I'm not very tech savvy so any clarifications about this would be appreciated! 

[image]
[image]

## Hardware/software information
Windows Version: 10

Tablet Driver version: TabletDriverV0.0.13 

Tablet model: Wacom Intuos CTH-480

freakode, 2018-03-15:
The high report rate comes from the smoothing filter;  if you turn this off your tablet will show a report rate of 133Hz.

SuperTilted, 2018-03-15:
Oh alright, I'm just gonna assume that its probably safe to leave that on then as well. Thanks!

closed 2018-03-15 by SuperTilted