hawku/TabletDriver #29: Add vertical/horizontal flipping

Milkitic, 2018-03-15:
Something like wacom's left hand mode for special use

Milkitic, 2018-03-15:
Sorry it's not wacom's features but it's still useful when you flip the screen at same time (retrying some map)

SuperTilted, 2018-03-15:
https://www.reddit.com/r/osugame/comments/83hbiq/testing_different_tablet_drivers_with_a_3d_printer/dvj6mco/

Is this what you're talking about? You should be able to invert the area by inputting 180 deg under rotation.

kevinwlau, 2018-03-15:
Duplicate of #20

Shoko84, 2018-03-15:
I understand it as a "mirroring" thing, is that what you want ?

Milkitic, 2018-03-15:
I help someone asked you to add a feature like Wacom's left hand mode (180 deg), but i forgot it's just a way to rotation, thinking it's something like mirroring the area and added an issue here. After I posted I found the truth but thinking "mirroring" thing can be used anyway: mirror the screen output at same time in some cases. Anyway sorry for misunderstand.

hawku, 2018-03-16:
Duplicate of #20 

[Frequently Asked Questions](https://github.com/hawku/TabletDriver/wiki/FAQ)

hawku, 2018-03-16:
[Frequently Asked Questions](https://github.com/hawku/TabletDriver/wiki/FAQ)

closed 2018-03-16 by hawku