hawku/TabletDriver #30: Near impossible to tap on CTH-490

SocksGalore, 2018-03-15:
So I have the driver installed and for the moment is seems to be working mostly properly, the only issue I've experienced is that I cannot click on anything either within osu or in other applications. I can however click and drag, either to scroll through the song list or to highlight text and stuff. My guess as to why this is happening is that the driver is counting all of my taps as drags, this was an issue to a smaller extent with the default Wacom driver, where I would have to keep my hand steady when clicking on things, especially outside of the game, with this one however no matter how much i try to steady my hand it still counts it as a drag. I have managed to very carefully, keeping the pen entirely vertical, gotten it to just click, which is another reason why I think this is the issue. 

If this is just a setting for me to change than I will do that, I can also provide more information as needed, I can get around this situation by using the track pad on my laptop (all my usb ports are full playing osu so I cant plug in a mouse) so there is no big rush from me, its just a bit of a pain to switch between them.

## Hardware/software information
Windows Version: 
Windows 10

Tablet Driver version: 
V0.0.14

Tablet model: 
Intuos CTH-490

SocksGalore, 2018-03-15:
Actually I may have been too hasty, playing around with it it seems like I can get it to work if I just press harder, not sure why I didn't think of that, probably because I didn't need too with the old driver, but yeah, pressing harder when I click seems to work, though it still works better in game than out, that was the case with the last driver too. I'll go ahead and close the thread.

closed 2018-03-15 by SocksGalore

hawku, 2018-03-15:
@SocksGalore Tip click pressure is forced to a higher value on Wacom 490 tablets.
Pen tip pressure doesn't react fast enough when you lift the pen off the tablet, so the click detection pressure is set to a higher number to help with that problem.

You can change the click detection pressure to zero by typing this in the to one of the "Commands" tab's fields and click Apply:
`ClickPressure 0`

Current default value for that is 500.