hawku/TabletDriver #31: Cursor suddenly clips to the edge of the screen.

nerrowcool, 2018-03-15:
Video:
[https://puu.sh/zHXU2/c572826d73.mp4](https://puu.sh/zHXU2/c572826d73.mp4)
[https://puu.sh/zHYb0/9872c3367d.mp4](https://puu.sh/zHYb0/9872c3367d.mp4)

Setting:
[image]

As you see, the cursor suddenly "teleported" to the edge of my screen after it reached a certain position and sometimes stuck at there for a while. It causes inconvenience and greatly affects my daily use of tablet.

## Hardware/software information
Windows Version: Windows 10

Tablet Driver version: 0.0.14

Tablet model: CTH-480

nerrowcool, 2018-03-15:
This situation still happens in other area setting.

nerrowcool, 2018-03-15:
Also, the wacom area setting automatically changed **"Top: 0, Bottom: 7335, Left: 0, Right: 13040"** to **"Top: 1, Bottom: 7336, Left: 0, Right: 13040"** while 13040/7335 is 16/9.

ErzaScarlettttt, 2018-03-26:
try minimizing the gui instead of closing it

hawku, 2018-04-09:
@nerrowcool Does that happen with the official Wacom drivers?

closed 2018-08-12 by hawku