hawku/TabletDriver #32: Tablet not found, I had never installed any wacom drivers on my computer

salmanmkc, 2018-03-15:
Tablet not found, I had never installed any wacom drivers on my computer, even removed Wacom tablet from device manager and it doesn't work. I did a windows reset, didn't work. Latest version of windows 10. Wacom Intous Ctl-480. I've even tried removing everything from the config and leaving just the CTL-480 bits. Still didn't work. Please help.

2018-03-15 06:26:42 [INFO] >> MaxY 13500
2018-03-15 06:26:42 [INFO] >> MaxPressure 2047
2018-03-15 06:26:42 [INFO] >> Width 216.000
2018-03-15 06:26:42 [INFO] >> Height 135.000
2018-03-15 06:26:42 [INFO] >> InitFeature 0x02 0x02
2018-03-15 06:26:42 [INFO] >> #
2018-03-15 06:26:42 [INFO] >> # Wacom CTL-680
2018-03-15 06:26:42 [INFO] >> #
2018-03-15 06:26:42 [INFO] >> Tablet 0x056A 0x0323 0x000D 0x0001
2018-03-15 06:26:42 [ERROR] Can't open HID tablet 0x056A 0x0323 0x000D 0x0001
2018-03-15 06:26:42 [INFO] >> Tablet 0x056A 0x0323 0xFF0D 0x0001
2018-03-15 06:26:42 [ERROR] Can't open HID tablet 0x056A 0x0323 0xFF0D 0x0001
2018-03-15 06:26:42 [INFO] >> Name "Wacom CTL-680"
2018-03-15 06:26:42 [INFO] >> ReportLength 10
2018-03-15 06:26:42 [INFO] >> ButtonMask 0x40
2018-03-15 06:26:42 [INFO] >> MaxX 21600
2018-03-15 06:26:42 [INFO] >> MaxY 13500
2018-03-15 06:26:42 [INFO] >> MaxPressure 1023
2018-03-15 06:26:42 [INFO] >> Width 216.000
2018-03-15 06:26:42 [INFO] >> Height 135.000
2018-03-15 06:26:42 [INFO] >> InitFeature 0x02 0x02
2018-03-15 06:26:42 [INFO] >> #
2018-03-15 06:26:42 [INFO] >> # Wacom CTL-690
2018-03-15 06:26:42 [INFO] >> #
2018-03-15 06:26:42 [INFO] >> Tablet 0x056A 0x033D 0x000D 0x0001
2018-03-15 06:26:42 [ERROR] Can't open HID tablet 0x056A 0x033D 0x000D 0x0001
2018-03-15 06:26:42 [INFO] >> Tablet 0x056A 0x033D 0xFF0D 0x0001
2018-03-15 06:26:42 [ERROR] Can't open HID tablet 0x056A 0x033D 0xFF0D 0x0001
2018-03-15 06:26:42 [INFO] >> Name "Wacom CTL-690"
2018-03-15 06:26:42 [INFO] >> ReportLength 10
2018-03-15 06:26:42 [INFO] >> ButtonMask 0x40
2018-03-15 06:26:42 [INFO] >> MaxX 21600
2018-03-15 06:26:42 [INFO] >> MaxY 13500
2018-03-15 06:26:42 [INFO] >> MaxPressure 2047
2018-03-15 06:26:42 [INFO] >> Width 216.000
2018-03-15 06:26:42 [INFO] >> Height 135.000
2018-03-15 06:26:42 [INFO] >> InitFeature 0x02 0x02
2018-03-15 06:26:42 [INFO] >> Type WacomIntuos
2018-03-15 06:26:42 [INFO] >> #
2018-03-15 06:26:42 [INFO] >> # Wacom CTH-460
2018-03-15 06:26:42 [INFO] >> #
2018-03-15 06:26:42 [INFO] >> Tablet 0x056A 0x00D1 0x000D 0x0001
2018-03-15 06:26:42 [ERROR] Can't open HID tablet 0x056A 0x00D1 0x000D 0x0001
2018-03-15 06:26:42 [INFO] >> Tablet 0x056A 0x00D1 0xFF0D 0x0001
2018-03-15 06:26:42 [ERROR] Can't open HID tablet 0x056A 0x00D1 0xFF0D 0x0001
2018-03-15 06:26:42 [INFO] >> Name "Wacom CTH-460"
2018-03-15 06:26:42 [INFO] >> ReportLength 10
2018-03-15 06:26:42 [INFO] >> ButtonMask 0x40
2018-03-15 06:26:42 [INFO] >> MaxX 14720
2018-03-15 06:26:42 [INFO] >> MaxY 9200
2018-03-15 06:26:42 [INFO] >> MaxPressure 1023
2018-03-15 06:26:42 [INFO] >> Width 147.200
2018-03-15 06:26:42 [INFO] >> Height 92.000
2018-03-15 06:26:42 [INFO] >> InitFeature 0x02 0x02
2018-03-15 06:26:42 [INFO] >> #
2018-03-15 06:26:42 [INFO] >> # Wacom CTH-460(A)
2018-03-15 06:26:42 [INFO] >> #
2018-03-15 06:26:42 [INFO] >> Tablet 0x056A 0x00D6 0x000D 0x0001
2018-03-15 06:26:42 [ERROR] Can't open HID tablet 0x056A 0x00D6 0x000D 0x0001
2018-03-15 06:26:42 [INFO] >> Tablet 0x056A 0x00D6 0xFF0D 0x0001
2018-03-15 06:26:42 [ERROR] Can't open HID tablet 0x056A 0x00D6 0xFF0D 0x0001
2018-03-15 06:26:42 [INFO] >> Name "Wacom CTH-460(A)"
2018-03-15 06:26:42 [INFO] >> ReportLength 10
2018-03-15 06:26:42 [INFO] >> ButtonMask 0x40
2018-03-15 06:26:42 [INFO] >> MaxX 14720
2018-03-15 06:26:42 [INFO] >> MaxY 9200
2018-03-15 06:26:42 [INFO] >> MaxPressure 1023
2018-03-15 06:26:42 [INFO] >> Width 147.200
[... 214 lines cut ...]
2018-03-15 06:26:42 [INFO] >> #  - CTH-461
2018-03-15 06:26:42 [INFO] >> #  - CTH-461(A)
2018-03-15 06:26:42 [INFO] >> #  - CTH-461SE
2018-03-15 06:26:42 [INFO] >> #  - CTH-470
2018-03-15 06:26:42 [INFO] >> #  - CTH-490
2018-03-15 06:26:42 [INFO] >> #  - CTH-661
2018-03-15 06:26:42 [INFO] >> #  - CTH-661(A)
2018-03-15 06:26:42 [INFO] >> #  - CTH-661SE
2018-03-15 06:26:42 [INFO] >> #  - CTH-480
2018-03-15 06:26:42 [INFO] >> #  - CTH-670
2018-03-15 06:26:42 [INFO] >> #  - CTH-680
2018-03-15 06:26:42 [INFO] >> #  - CTH-690
2018-03-15 06:26:42 [INFO] / End of 'config\wacom.cfg'
2018-03-15 06:26:42 [INFO] >> CheckTablet
2018-03-15 06:26:42 [ERROR] Tablet not found!
2018-03-15 06:26:42 [ERROR] Check the list of supported tablets from the GitHub page.
2018-03-15 06:26:42 [ERROR] http://github.com/hawku/TabletDriver
Driver stopped. Restarting!

## Hardware/software information
Windows Version: 

Tablet Driver version: 

Tablet model:

Shoko84, 2018-03-15:
Do you have the Fall Creators Update ? We've seen yesterday that there may be some issues at finding the tablet if don't have this version.

Also : 
> Add these lines to the Commands tab's first field and press Apply:
> 
> - log debug.txt
> - debug 1
> 
> That should create a debug.txt log file which contains lots of tablet status information.
> Try to reproduce the issue when the debug log is running.
> Log file will be huge if you play too long with logging enabled, but the log file will be cleared every time you apply or save settings.
> When you have captured the issue, close the GUI without saving or applying, zip the log file and upload it here.

May help about finding the problem !

hawku, 2018-03-15:
@Shoko84 That debug log doesn't help when the tablet is not found. Startuplog.txt might give some information.

@salmanmkc Try to run the TabletDriverGUI as administrator.
Can you find Wacom's HID device from the Windows Device Manager?
http://hwk.fi/TabletDriver/WacomHID.png
Device instance path should start with "HID\VID_056A"

salmanmkc, 2018-03-16:
https://i.imgur.com/aY4c0ly.png

salmanmkc, 2018-03-16:
https://i.imgur.com/XcnZnIN.png

This is what i'm getting now, I have that in my device manager: 
https://i.imgur.com/XYFsoJk.png

Shoko84, 2018-03-16:
By any chance, did you separate the `TabletDriverGUI.exe` outside of its original folder when you're launching it ?

salmanmkc, 2018-03-16:
https://i.imgur.com/UjkkcfI.png

salmanmkc, 2018-03-16:
Reinstalled again, did everything again and getting the old error:
https://i.imgur.com/Z79ctFI.png

salmanmkc, 2018-03-16:
okay so I plugged it back in to get it back in device manager after installing the drivers whilst I manually had deleted the drivers that auto get installed when plugging it in, ran as admin and it worked. Thanks guys! Great support :)

salmanmkc, 2018-03-16:
When playing osu! should I tick raw input with these drivers?

salmanmkc, 2018-03-16:
Should I keep raw input (osu setting) on with these drivers or off?

hawku, 2018-03-16:
Raw input enabled and disable the "Map absolute raw input to the osu! window"

salmanmkc, 2018-03-17:
Problem is back... It crashed and I tried reopen it with this error: https://i.imgur.com/MdpMlzv.png

salmanmkc, 2018-03-17:
Upon checking for the tabletdriverservice in the bin folder, it has now disappeared

salmanmkc, 2018-03-17:
Reinstalled with the new version and it works now. I have no idea why the file keeps getting deleted, I don't have anti-virus on

closed 2018-03-26 by hawku