hawku/TabletDriver #33: Implement button tablet handling

Shoko84, 2018-03-15:
As a additional feature, it could be a good idea to implement the button handling of tablets

I don't think it's a hard feature and not necessarily a priority with all these tablet recognization issues but I may work on it if you want (at least for the CTL480)

hawku, 2018-03-15:
I was working on that before I posted the driver to Reddit.

To properly implement the feature, you would need to create a new thread that handles keyboard/mouse macros.
All that also needs some user interface changes, commands, testing and problem solving with different tablets.

I've added a second HID device to the Tablet object:
https://github.com/hawku/TabletDriver/blob/master/TabletDriverService/ProcessCommand.cpp#L89

I doesn't have any functionality, but I was planning to use that to read the tablet's HID interface that sends the button and touch states.
Use USBlyzer or other USB capture software to figure out which interface sends the button states.
Some tablets might send the button states on the same interface as pen position.

Shoko84, 2018-03-15:
Gotcha' !

Do you mind if you can create a new branch about this particular issue, since we can work on that together + with some other peoples that are interested ? 😄 

I've some knowledges with WPF applications so I might start on the GUI interface firstly !

hawku, 2018-03-16:
@Shoko84 I will create a branch "buttons" after I have committed and pushed latest changes.

Shoko84, 2018-03-16:
Perfect, I will get on it soon, thanks !

Shoko84, 2018-03-16:
I don't find anything about getting tablet informations _(especially the tablet model)_ from GUI driver, is there anything to catch this somewhere ?

hawku, 2018-03-16:
Currently the tablet name detection is just a quick'n'dirty hack:
https://github.com/hawku/TabletDriver/blob/master/TabletDriverGUI/MainWindow.xaml.cs#L1339

That code is parsing the tablet name from the TabletDriverService console output.
I need to figure out a better way to transfer the status information between the DriverService and GUI.

I may add a new logging level which is only used in the status information communication, but for now the quick'n'dirty hack will do.

hawku, 2018-03-16:
@Shoko84 I've added the "buttons" branch.

Shoko84, 2018-03-17:
So the GUI part is mostly done in my local work but I'm struggling at triggering virtual keyboard shortcuts _(similar to what Wacom drivers have)_.

I wanted to add this feature to give tablet buttons an useful utility, at least more than just simulate mouse buttons.
Everything I'm trying isn't working as intended so I'm quite sad. I think I will leave this besides for now and work just on mapping tablet buttons with mouse clicks thanks to your second HID first implmentation.

By the way, since I just own a CTL-480, I might need some help about which informations are sent with other devices (if there are now way to emulate devices)

hawku, 2018-03-18:
@Shoko84  It is possible to create virtual mouse/keyboard actions with this: https://msdn.microsoft.com/en-us/library/windows/desktop/ms646310(v=vs.85).aspx

That function should be run on a separate thread, because i think that it will block until the keyboard action is done.
Normal tablet actions are currently running in this thread: https://github.com/hawku/TabletDriver/blob/63ddbd7d3fff0d41293b3034741691970dd9c677/TabletDriverService/Main.cpp#L44-L162

Virtual keyboard actions should be executed from a new thread that reads the input from the secondary HID device. Keyboard macros should be stored in the Tablet object and that will also need a console command for setting up those macros.

Currently there isn't any tablet emulation, but I'm planning to make HID and WinUSB device emulators, so it would be possible to test the driver in a virtual machine.

Shoko84, 2018-03-21:
Hi back, the GUI part is almost done (except saving shortcuts into the config file, but it's not a hard work tbh) but I have a issue driver-side when getting the new command (TabletMacro)

To be short, the TabletMacro command receive a string like `"0,4,47;1,2,4,59"` which means that, for the ButtonMap at the index `0`, the ASCII keymacro 4 and 47 are pressed (and then released _code-side_) and keys 2,4 and597 are pressed for the ButtonMap[1]. Also the value into ButtonMap indexes for shortcuts will be a different value to differentiate these (_6_ iirc)

The thing is here, I do catch the command (goes into _else if_ correctly) but when I try to _LOG_INFO_ the _cmd->values_ array, it looks like the values are corrupted or not correctly initialized. But if it was really the case, then why it does enter in the condition ? 🤔 

The _else if_ condition code written is : 

```cpp
...
        //
	// Tablet buttons macro
	//
	else if (cmd->is("TabletMacro")) {
		LOG_INFO("Entering TabletMacro command\n");
		if (!CheckTablet()) return true;
		LOG_INFO("ValueCount: %d\n", cmd->valueCount);

		std::vector<std::string> v;

		for (auto it : cmd->values)
			v.push_back(std::string(it));

		LOG_INFO("Cmd sent: %s\n", Utils::join(v, " "));

		std::vector<std::string> tempSplit = Utils::split(Utils::join(v, " "), ';');
		std::vector<std::vector<std::string>> tabletMacros;

		for (auto &it : tempSplit)
			tabletMacros.push_back(Utils::split(it, ','));

		for (auto &it : tabletMacros)
		{
			for (auto it2 : it)
				LOG_INFO("%s\n", it2);
			LOG_INFO("\n\n");
			//std::vector<int> keysMacro;

			/*std::transform(it.begin() + 1, it.end(), keysMacro.begin(), [](const std::string& value)
			{
				return std::stoi(value);
			});*/
			//tablet->buttonTabletMap[std::stoi(it[0])] = keysMacro;
			//LOG_INFO("Set the macro to the tablet button [%d]\n", std::stoi(it[0]));
		}
	}
...
```

And here is what it's been written in the console : 
https://puu.sh/zMmMd/b6723b86ea.png

I'm a bit confused and stucked, but if I can understand why and how to fix it, it should be almost finished to be sure 😄 

I can push my local work either on my fork repository or into the _branches_ branch here if needed !

hawku, 2018-03-21:
@Shoko84 LOG_INFO uses the snprintf function to format the string and that function doesn't support C++ strings.
You have to use `string.c_str()` to get the C-style string for the logging commands.

Example:
```
LOG_INFO(">> %s\n", cmd->line.c_str());
```

Shoko84, 2018-03-21:
Oh fu*k yeah obviously I feel dumb lmao, I will try that thanks

Shoko84, 2018-03-29:
Okay so I'll look forward at your changes with HIDDevice, got busy and will have time this week-end, I'm being optimistic to finish that really soon, thanks for your help!

Shoko84, 2018-04-07:
My free time is really bad at the moment tbh, I wish I can finish that asap but can't. Will try to give it a look tomorrow / tonight..

If anyone wanna try to work on it, the last version is pushed and available on the `buttons` branch iirc

hawku, 2018-04-09:
@Shoko84 No worries. I don't have much of free time either.
I will look in to the button support when I have the time.

hawku, 2019-01-09:
Button handling was implemented in version 0.2.2

closed 2019-01-09 by hawku