hawku/TabletDriver #35: [ERROR] Tablet not found!

Varukiri, 2018-03-15:
## Hardware/software information

Windows Version: 7 x64

Tablet Driver version: your driver 0.14

Tablet model: g430

## Issue

I installed driver from main folder:
[image]

Then when I start TabletDriverGUI this happens:
[image]

I also tried driver from folder "driver_vmulti" - same issue

Tried to install https://www.xp-pen.com/upload/download/20180130/XP-PenWin(20180130).zip
but this driver doesn't work. It always says device not detected. Only one that worked for me was g430/star04 old driver.

Here is startuplog file: [startuplog.txt](https://github.com/hawku/TabletDriver/files/1815378/startuplog.txt)

Varukiri, 2018-03-15:
after putting cfg files in same folder with TabletDriverService this happened
[debug.txt](https://github.com/hawku/TabletDriver/files/1815483/debug.txt)

Varukiri, 2018-03-15:
from log
2018-03-15 15:44:48 [INFO] >> #
2018-03-15 15:44:48 [INFO] >> # XP Pen G430
2018-03-15 15:44:48 [INFO] >> #
2018-03-15 15:44:48 [INFO] >> Tablet 0x28BD 0x0075 0xFF0A 0x0001
2018-03-15 15:44:48 [ERROR] Can't open HID tablet 0x28BD 0x0075 0xFF0A 0x0001
2018-03-15 15:44:48 [INFO] >> Name "XP Pen G430"
2018-03-15 15:44:48 [INFO] >> ReportId 0x02
2018-03-15 15:44:48 [INFO] >> ReportLength 8
2018-03-15 15:44:48 [INFO] >> ButtonMask 0x80
2018-03-15 15:44:48 [INFO] >> MaxX 45720
2018-03-15 15:44:48 [INFO] >> MaxY 29210
2018-03-15 15:44:48 [INFO] >> MaxPressure 2047
2018-03-15 15:44:48 [INFO] >> Width 102.4
2018-03-15 15:44:48 [INFO] >> Height 76.2
2018-03-15 15:44:48 [INFO] >> InitReport 0x02 0xB0 0x02 0x00 0x00 0x00 0x00 0x00

hawku, 2018-03-15:
@Varukiri Try to run the TabletDriverGUI as an administrator.

Are you sure that your tablet is G430 and not the newer model G430**S**?

Can you find tablet's HID device from the Windows Device Manager?
http://hwk.fi/TabletDriver/WacomHID.png, but the device instance path should start with `HID\VID_28BD`

Varukiri, 2018-03-15:
@hawku
 I am always run it as an administrator.

 My tablet is exactly G430. 

There is so many maybe because I used some different USB ports for tablet.
[image]

hawku, 2018-03-16:
Did you check if you have other tablet driver services or processes running?
For example XP-Pen's PentabletService.exe or Pentablet.exe

eternalstandstill, 2018-03-16:
yeah, i uninstalled xp pen drivers, and don't see them in the windows task manager

hawku, 2018-03-16:
Something weird is going on. It looks like your tablet isn't same device as my G430.
Your tablet is missing the driver communication interface (0xFF0A)

My G430:
```
[DEBUG] HID Device: Vendor: 'UGTABLET' Product: 'TABLET G3 4x3'
[DEBUG]   Usage Page: 0x0001, Usage: 0x0000
[DEBUG]   Usage Page: 0x000D, Usage: 0x000E
[DEBUG]   Usage Page: 0xFF0A, Usage: 0x0001
```
Your G430:
```
[DEBUG] HID Device: Vendor: 'UGTABLET' Product: 'TABLET G3 4x3'
[DEBUG]   Usage Page: 0x0001, Usage: 0x0000
[DEBUG]   Usage Page: 0x000D, Usage: 0x0001
[DEBUG]   Usage Page: 0x000D, Usage: 0x000E
```

Does the latest XP-Pen drivers work with your tablet?
https://www.xp-pen.com/upload/download/20180130/XP-PenWin(20180130).zip

Varukiri, 2018-03-16:
Only old g430/star04 driver works with my g430 but I'm sure that this is exactly 430. I've seen many 430 users had error on new driver: "device not detected". So your 430 works with the latest??

hawku, 2018-03-16:
@Varukiri 
Yes, my G430 works with the latest XP-Pen drivers.
I don't understand why XP-Pen has made two different G430 tablets with the same USB Product ID.
XP-Pen G540 might also share the same USB Product ID.

Varukiri, 2018-03-16:
I bought my July 2016 btw

hawku, 2018-03-16:
I bought mine about 2 months ago from this seller: https://www.aliexpress.com/item//32609175289.html

I can see the product listing history from AliExpress and it has this text on the product page: "Model Number: StarG430_**B**"

dnsklnv, 2018-03-22:
Same problem. I bought my ~march 2016. The only working driver is star04 driver. 
Otherwise tablet itself works just fine. I think you should mention this problem in FAQ to save time for others.

Syntractrix, 2018-03-25:
I too have a g430, and something similar occurs while trying to run these drivers.
I'll upload the log generated by it: 
[startuplog.txt](https://github.com/hawku/TabletDriver/files/1844985/startuplog.txt)
GUI Console generated error:
[image]
Here are all of the drivers generated by the tablet in device manger:
[image]
[image]
This is regular g430 not the g430s (newer drivers dont work on it)
OS: W7 x64
[image]

hawku, 2018-03-25:
@Syntractrix That is the old model.

Reddit user CaptainShrimps sent me some USB data from the older G430 model and it looks like it is almost impossible to support that tablet.

XP-Pen have changed the way they do the USB communication on the newer models. That makes it very hard for this driver or XP-Pen's latest drivers to support the older devices.

bobross12, 2018-03-29:
Is there going to be a driver for the Star g430s? Just wondering because i hate trying to map the area with xp pen driver

closed 2018-04-01 by hawku

Opera-its-browser, 2019-12-06:
Hello, I have a tablet XP-pen g430s, and when I turn on your driver comes out error, " Driver Stopped, Restarting! Check console!!!". I bought it two or three weeks ago. What should I do?
[startuplog.txt](https://github.com/hawku/TabletDriver/files/3931655/startuplog.txt)

Kirasabi, 2020-02-24:
Yo, i have tablet xp-pen star g960. Is it possible to make it work on tablet driver GUI?

kaspishehe, 2020-10-07:
Hello, i have similar problem with my xp pen g640 like the first person on this thread. Tablet not found and driver stopped.
I installed vmulti and deleted previous drivers,

[startuplog.txt](https://github.com/hawku/TabletDriver/files/5340600/startuplog.txt)
[vmulti log.txt](https://github.com/hawku/TabletDriver/files/5340602/vmulti.log.txt)

Duckhat9, 2021-08-06:
I am using Veikk S640 tablet. I have install your drivers.  Tried everything but it doesn't work.
[image]