hawku/TabletDriver #36: QUESTION: How to export area?

Scotxd, 2018-03-15:
Not an issue, but how do I export the area from my old wacom 5.3.5-3 drivers to these?

## Hardware/software information
Windows Version: win 10

Tablet Driver version: 0.0.13

Tablet model: ctl-471

SuperTilted, 2018-03-15:
Don't think there's a way for exporting yet but you can just manually input the values of your old area yourself by clicking on the "Wacom Area" right under Rotation. A pop up will show up and allow you to input your values in the same format as a Wacom driver's. 

[image]

hawku, 2018-03-16:
@Scotxd You could try this: https://github.com/SaltimusPrime/WacomAreaChanger/releases

I might add way to import Wacom backup file.

closed 2018-03-16 by hawku