hawku/TabletDriver #37: CTL-490 Default Click Pressure Increased

FullTablet, 2018-03-15:
The drivers here set the ClickPressure to 500 by default, while with the official drivers the pressure was 0. This makes the tablet feel awkward to use, and many users might not know how to set it like it was before.

Setting a "Command After settings" sets it to how it was before.
ClickPressure 0

## Hardware/software information
Windows Version: Windows 7 64-bit

Tablet Driver version: 0.0.13

Tablet model: CTL-490

hawku, 2018-03-16:
@FullTablet That default click pressure is a workaround for the slow/filtered pen tip pressure change when you lift the pen off the tablet surface. 

I will remove that workaround by setting the values back to 0. I'll have to figure out better solution for that problem.

hawku, 2018-03-16:
Wacom 490 ClickPressure settings removed in V0.0.15
Commit: e734020f452030bc0dcc0e8d446f3cb311aefcec

closed 2018-03-16 by hawku