hawku/TabletDriver #42: Support for XPPen G540

lolify, 2018-03-17:
My tablet is the G540 and it doesn't say it's supported, the driver identifies it as the G430, though changing settings in the driver doesn't seem to work.
I tried to get it working myself, VID seems to be 0x28BD and PID 0x0075 but I have no idea how to correctly implement them in the `tablet.cfg`.
Would be awesome if it's possible to add support!

0101zyga, 2018-03-20:
I hope my G540 gets support for these drivers!

Donce114, 2018-03-27:
Please get G540 drivers out soon :(

hawku, 2018-03-27:
@Donce114 I've ordered the G540 from China, so it might take a few weeks for it to arrive.

0101zyga, 2018-03-27:
I downloaded drivers and they seem to detect my G540 as G430, basically I just changed the area size to match my G540 and everything is working fine

hawku, 2018-12-10:
@lolify @0101zyga @Donce114 Please, check the issue #301

closed 2019-01-09 by hawku