hawku/TabletDriver #43: Cursor and Pen problem

Zmatureeno, 2018-03-17:
When I click on 2 buttons on stylus(my thumb is holding them while playing) my cursor glitches and appears twice on my monitor and also bugs a lot by teleporting from circles, I tried to disable them, didnt work. 

## Hardware/software information
Windows Version: windows 7

Tablet Driver version: 0.0.16

Tablet model: ctl480,but using 471 pen.

hawku, 2018-03-17:
I would like to have more information about the problem:
1. Update to the latest version of the driver.
2. Open the Console tab and right click the console output to open a context menu.
3. Select "Start Debug Log" from that menu. It will start to record debug data to `debug_*.txt` file in the driver's folder.
4. Try to reproduce the problematic pen movement while the debug logging is running.
5. Close the GUI when you are sure that you have captured the problem.
6. Attach the `debug_*.txt` and `startuplog.txt` files from the driver's folder to a GitHub issue comment.

Zmatureeno, 2018-03-18:
[debug_2018-03-18_01_28_29.txt](https://github.com/hawku/TabletDriver/files/1822214/debug_2018-03-18_01_28_29.txt)
[startuplog.txt](https://github.com/hawku/TabletDriver/files/1822215/startuplog.txt)

So here they are, I caught plenty of those bugs so I hope it will be enought. Also I just realized it happens ONLY in osu! itself, not in windows which is unlucky.

hawku, 2018-03-18:
The problem was that there was another TabletDriverService.exe running in the background.
Added multi-instance prevention in the commit ee119af5fe13a234213fb49d07b699b3a10b9bdf

closed 2018-03-18 by hawku