hawku/TabletDriver #44: Clicking with the tablet.

Panchosauriooo, 2018-03-17:
When i click with the pen nib, it holds the click for a few milliseconds( <0.15s) resulting in unintended clicks.

## Hardware/software information
Windows Version: Windows 10

Tablet Driver version: v0.0.17

Tablet model: CTL-490

Panchosauriooo, 2018-03-17:
Additional to this, it only happens when you hover at a small height(almost touching the tablet)

hawku, 2018-03-17:
Does that happen with Wacom drivers?

Panchosauriooo, 2018-03-18:
No, and neither does it in the 0.0.13 version since thats the version from which i updated. I will try with the other versions.

Panchosauriooo, 2018-03-18:
Happens from version 0.0.15 and up

FullTablet, 2018-03-18:
Might be related to this: https://github.com/hawku/TabletDriver/issues/37

Is your pen tip sticky? Try removing the tip with the nib removal tool at the back of the tablet, and check and clean it before inserting again.

Otherwise, there might be some hardware difference (mine works with no issue ClickPressure 0), you could try to use a higher click pressure, but that way you would need to press the tip of the pen to click instead of just touching the surface.

hawku, 2018-03-18:
@Panchosauriooo Try the latest version (http://hwk.fi/TabletDriver/TabletDriverV0.0.18.zip)
I've added an another fix for the 490 clicking problem.

@FullTablet It seems that it's a hardware problem. I checked how Wacom drivers fix that problem and implemented a similar solution: e5745204f8a35823c63492b326b6e3b9456aa0c1

Panchosauriooo, 2018-03-18:
@hawku version 0.0.18 does indeed fix it, Thanks.

closed 2018-03-18 by Panchosauriooo