hawku/TabletDriver #48: [ERROR] Tablet init failed! (Cant find CTL-480 ?

marcoboy123, 2018-03-18:
`2018-03-18 17:04:39 [INFO] \ Reading 'config/init.cfg'
2018-03-18 17:04:39 [INFO] >> Log startuplog.txt
2018-03-18 17:04:39 [INFO] Log file 'startuplog.txt' opened.
2018-03-18 17:04:39 [INFO] >> LogDirect true
2018-03-18 17:04:39 [INFO] Log direct print = True
2018-03-18 17:04:39 [INFO] >> Include "config\tablet.cfg"
2018-03-18 17:04:39 [INFO] \ Reading 'config\tablet.cfg'
2018-03-18 17:04:39 [INFO] >> #
2018-03-18 17:04:39 [INFO] >> # Wacom CTL-470
2018-03-18 17:04:39 [INFO] >> #
2018-03-18 17:04:39 [INFO] >> # VID: 0x056a
2018-03-18 17:04:39 [INFO] >> # PID: 0x00dd
2018-03-18 17:04:39 [INFO] >> # HID Usage Page: 0x0D
2018-03-18 17:04:39 [INFO] >> # HID Usage: 0x01
2018-03-18 17:04:39 [INFO] >> #
2018-03-18 17:04:39 [INFO] >> Tablet 0x056a 0x00dd 0x0D 0x01
2018-03-18 17:04:39 [WARNING] Can't open HID tablet 0x056A 0x00DD 0x000D 0x0001
2018-03-18 17:04:39 [INFO] >> Name "Wacom CTL-470"
2018-03-18 17:04:39 [INFO] >> ReportId 0x02
2018-03-18 17:04:39 [INFO] >> ReportLength 10
2018-03-18 17:04:39 [INFO] >> ButtonMask 0x40
2018-03-18 17:04:39 [INFO] >> MaxX 14720
2018-03-18 17:04:39 [INFO] >> MaxY 9200
2018-03-18 17:04:39 [INFO] >> MaxPressure 1023
2018-03-18 17:04:39 [INFO] >> Width 147.2
2018-03-18 17:04:39 [INFO] >> Height 92.0
2018-03-18 17:04:39 [INFO] >> InitFeature 0x02 0x02
2018-03-18 17:04:39 [INFO] >> #
2018-03-18 17:04:39 [INFO] >> # Wacom CTH-470 ???
2018-03-18 17:04:39 [INFO] >> #
2018-03-18 17:04:39 [INFO] >> Tablet 0x056a 0x00de 0x0D 0x01
2018-03-18 17:04:39 [WARNING] Can't open HID tablet 0x056A 0x00DE 0x000D 0x0001
2018-03-18 17:04:39 [INFO] >> Name "Wacom CTH-470"
2018-03-18 17:04:39 [INFO] >> ReportId 0x02
2018-03-18 17:04:39 [INFO] >> ReportLength 10
2018-03-18 17:04:39 [INFO] >> ButtonMask 0x40
2018-03-18 17:04:39 [INFO] >> MaxX 14720
2018-03-18 17:04:39 [INFO] >> MaxY 9200
2018-03-18 17:04:39 [INFO] >> MaxPressure 1023
2018-03-18 17:04:39 [INFO] >> Width 147.2
2018-03-18 17:04:39 [INFO] >> Height 92.0
2018-03-18 17:04:39 [INFO] >> InitFeature 0x02 0x02
2018-03-18 17:04:39 [INFO] >> #
2018-03-18 17:04:39 [INFO] >> # Wacom CTH-670 ???
2018-03-18 17:04:39 [INFO] >> #
2018-03-18 17:04:39 [INFO] >> Tablet 0x056a 0x00df 0x0D 0x01
2018-03-18 17:04:39 [WARNING] Can't open HID tablet 0x056A 0x00DF 0x000D 0x0001
2018-03-18 17:04:39 [INFO] >> Name "Wacom CTH-670"
2018-03-18 17:04:39 [INFO] >> ReportId 0x02
2018-03-18 17:04:39 [INFO] >> ReportLength 10
2018-03-18 17:04:39 [INFO] >> ButtonMask 0x40
2018-03-18 17:04:39 [INFO] >> MaxX 21648
2018-03-18 17:04:39 [INFO] >> MaxY 13700
2018-03-18 17:04:39 [INFO] >> MaxPressure 1023
2018-03-18 17:04:39 [INFO] >> Width 216.48
2018-03-18 17:04:39 [INFO] >> Height 137.0
2018-03-18 17:04:39 [INFO] >> InitFeature 0x02 0x02
2018-03-18 17:04:39 [INFO] >> #
2018-03-18 17:04:39 [INFO] >> # Wacom CTL-471
2018-03-18 17:04:39 [INFO] >> #
2018-03-18 17:04:39 [INFO] >> Tablet 0x056a 0x0300 0xFF0D 0x01
2018-03-18 17:04:39 [WARNING] Can't open HID tablet 0x056A 0x0300 0xFF0D 0x0001
2018-03-18 17:04:39 [INFO] >> Name "Wacom CTL-471"
2018-03-18 17:04:39 [INFO] >> ReportLength 10
2018-03-18 17:04:39 [INFO] >> ButtonMask 0x40
2018-03-18 17:04:39 [INFO] >> MaxX 14720
2018-03-18 17:04:39 [INFO] >> MaxY 9225
2018-03-18 17:04:39 [INFO] >> MaxPressure 1023
2018-03-18 17:04:39 [INFO] >> Width 147.2
2018-03-18 17:04:39 [INFO] >> Height 92.25
[... 550 lines cut ...]
2018-03-18 17:04:39 [INFO]   Button Mask = 0x40
2018-03-18 17:04:39 [INFO]   Button Map = 1 2 3 0 0 0 0 0 
2018-03-18 17:04:39 [INFO]   Tablet init feature report: { 0x02, 0x02 }
2018-03-18 17:04:39 [INFO] 
2018-03-18 17:04:39 [INFO] Area:
2018-03-18 17:04:39 [INFO]   Desktop = 3840px x 1080px
2018-03-18 17:04:39 [INFO]   Screen Map = 1920px x 1080px @ X+1920px, Y+0px
2018-03-18 17:04:39 [INFO]   Tablet =  48.91mm x 27.51mm @ Min(26.57mm, 28.76mm) Max(75.48mm, 56.27mm)
2018-03-18 17:04:39 [INFO]   Rotation matrix = [1.000000,-0.000000,0.000000,1.000000]
2018-03-18 17:04:39 [INFO] 
2018-03-18 17:04:39 [INFO] >> Start
2018-03-18 17:04:39 [ERROR] Tablet init failed!
2018-03-18 17:04:39 [ERROR] Possible fixes:
2018-03-18 17:04:39 [ERROR] 1) Uninstall other tablet drivers.
2018-03-18 17:04:39 [ERROR] 2) Stop other tablet driver services.
`

## Hardware/software information
Windows Version: Windows 10

Tablet Driver version: v.0.0.18 

Tablet model: CTL-480 Wacom

Yeah yeah you can see all my USB devices plugged in.. and idc :P

hawku, 2018-03-18:
Check the issue #1

marcoboy123, 2018-03-18:
So basically, update Windows 10 ?

closed 2018-03-18 by marcoboy123

hawku, 2018-03-18:
Yes, you only need to install the vmulti driver.