hawku/TabletDriver #49: Run Program on Startup (Minimized)

Frubbs, 2018-03-19:
Hello, this isn't an issue so much as a suggestion! I think it would be convenient if the program booted on start, minimized. I made a thread on Reddit for a temporary solution to this quality of life issue; but it would be much more streamlined to incorporate it into the program itself. 

I hope you consider it! Thank you for your time, you've concocted some phenomenal drivers and I want you to know that I, and I am sure many others are eternally grateful for them.

Is there somewhere I can donate to you? If so, message me on Reddit or comment in the thread linked below!

Temporary Solution Thread:
https://www.reddit.com/r/osugame/comments/85hv9u/guide_on_how_to_run_the_new_tablet_drivers_on/

## Hardware/software information
Windows Version: Windows 10 Pro (Release 1709)

Tablet Driver version: 0.0.18

Tablet model: Wacom CTL-480

hawku, 2018-03-19:
@Frubbs Thanks for the suggestion!

I added `-hide` command line argument when I saw your Reddit comment, but I'll still have to do some testing.

hawku, 2018-03-19:
@Frubbs I've added "Run at Windows startup". Check out the latest version.

closed 2018-11-29 by hawku