hawku/TabletDriver #52: More Keybind Options for Pen and Tablet Buttons?

Frubbs, 2018-03-19:
The biggest issue with switching from the Wacom Drivers to these is that I do a lot of PhotoShop work, so I've lost access to binding the buttons on the pen to keyboard functions like the space bar, as well as the 4 buttons on the top of the tablet have no use now. Any chance you'd implement more keybind options for the pen and tablet buttons?

Like I said in my last issue, I'd love to donate, where can I do so?

## Hardware/software information
Windows Version: 

Tablet Driver version: 

Tablet model:

hawku, 2018-03-19:
Check the issue #33 
@Shoko84 have made some progress with tablet button support, so maybe you could ask him about that.

What I will appreciate more than money is that people donate their time to look in the code, fix problems and help others at [GitHub issues](https://github.com/hawku/TabletDriver/issues)

I think that laws in my country prohibit asking and taking monetary donations without fundraising permit, so you shouldn't throw few cents to my PayPal mikko.jaakkola@gmail.com !

Shoko84, 2018-03-20:
Dunno where you're living but yeah, got some friends with that same issue so, be careful lol

Besides that, didn't got much time to work on it right now (school is a priority hehe), I'll try to finish that this week 😉

hawku, 2018-03-21:
Duplicate of #33

closed 2018-04-01 by hawku