hawku/TabletDriver #54: TabletDriverGUI.exe doesn't start anything

Zeny97, 2018-03-19:
As stated after executing the TabletDriverGUI.exe nothing happens. I followed all the steps according to the instructions with no luck whatsoever. Only thing that happens is a really short loading icon with nothing after that.

## Hardware/software information
Windows Version: Win 10 64Bit

Tablet Driver version: V0.1.0

Tablet model: CTL-490

Banbeucmas, 2018-03-20:
Hmm, can you try to reinstall all of the requirement program?

Zeny97, 2018-03-20:
I tried that out, but still the same problem.

hawku, 2018-03-20:
@Zeny97  I've modified the installation instructions. https://github.com/hawku/TabletDriver#installation

You might be missing the .NET Framework runtimes:
https://www.microsoft.com/net/download/dotnet-framework-runtime

Zeny97, 2018-03-20:
@hawku let me just say everything i did.

First of all i uninstalled both redist packages, then i ran the "remove_vmulti:driver.bat".
I made sure, that no wacom drivers were preinstalled.
Then i just restarted my computer for the sake of restarting, so i can follow the instructions as if i had nothing.

So i downloaded the redist packages x86 and x64 (allthough the x64 is now missing in the instructions) aswell as the .NET Framework version, which said that it's already on this PC during the setup, so that wasnt a problem.
after that i downloaded the TabletDriverV0.1.0 and unziped it directly onto my C harddrive (closest path)
i executed the "install_vmulti_driver.bat" and as you can see here with success http://puu.sh/zLDXM/0132567c93.png

However allthough everything seemed to be fine, nothing happens when i start the .exe 

I get the feeling the world doesn't want me to paly Osu! with tablet.
If you have any suggestions I'm more than glad hearing it.

Greetings

Zeny97, 2018-03-28:
@hawku So I tried getting it to work today and it finally did.
I changed 1 thing for TabletDriverGUI.exe
I went to the compatibility tab and checkmarked the box "Run this program in compatibilty mode for:" and put WIndows 7.
I use a 64bit Windows10 version, so i dont have any clue why it works after that. Just wanted to let you know

hawku, 2018-03-28:
@Zeny97 Thanks for figuring out a solution to the problem.

I don't remember anybody else having the same exact problem, but this solution might help others with similar problems.

closed 2018-04-01 by hawku