hawku/TabletDriver #56: Startup on Boot Doesn't Set High Affinity

Frubbs, 2018-03-20:
Hello! I was noticing after using the included startup on boot, that TabletDriverGUI.exe was no longer booting with high priority and I was experiencing some input lag for some reason. TabletDriverService boots with high priority, but it was the GUI priority that was causing the delay for some reason. Could you please make it so it boots into high priority? Thanks

## Hardware/software information
Windows Version: 

Tablet Driver version: 

Tablet model:

Frubbs, 2018-03-21:
Also, is it possible that the Tablet Driver is affecting my mouse movement? It seems my mouse is moving differently for some reason.

hawku, 2018-03-21:
The GUI process priority shouldn't affect the cursor movement. Does the lag get worse when you set the GUI process priority to "Low"?

How is your mouse moving differently after installing the driver?
Is the movement different when you start the GUI or uninstall the vmulti driver?

closed 2018-03-26 by hawku