hawku/TabletDriver #63: "Driver stopped. Restarting! Check console !!!"

temporaryrage, 2018-03-21:
Every time I would install huion_64 the tablet stops working. If I don't install it, it would say "Driver stopped. Restarting! Check console !!!". Help?

## Hardware/software information
Windows Version: 10

Tablet Driver version: Latest

Tablet model: Huion 420

vietnam13231, 2018-03-25:
Are you sure you deleted previous drivers? If yes, delete folders left by stock driver (configuration backups can stay, you can use them in this driver), install this driver, restart your PC. If it still doesn't work, then I'm lost...

Gerodar, 2018-03-26:
I have the same problem and I checked that I uninstalled every possible driver that I had. Program keeps saying "Driver stopped. Restarting! Check console !!!" and all listed fixes don't work.

Running on Win10 with latest drivers on a CTL-471

hawku, 2018-03-27:
@temporaryrage Upload the startuplog.txt to here.

@Gerodar Check the issue #1

closed 2018-08-12 by hawku