hawku/TabletDriver #74: ctl471 180 rotation doesnt match the lefthand mode area position

BakaFlame, 2018-03-25:
180 rotation doesnt match the lefthand mode area position...
it has the wrong position, i was using **top right** tablet area in lefthand mode, same position in td but i have to move it to **bottom left** and do 180 rotation.
wacom driver cursor is lower than the td one, please fix this, it looks werid lol

and i thought the screen map section left & top offset can fix that problem, but it just changed my tablet position, i cant reach the edge of my screen...

## Hardware/software information
Windows Version: windows10 64bits

Tablet Driver version: 5.3-5-3

Tablet model: CTL471

hawku, 2018-03-25:
Is that really that hard to understand?
180 degree rotation rotates the tablet area, not the tablet.
Tablet area arrow show the direction of the tablet area.

I will add "Left handed (Invert)" option in the next version.

hawku, 2018-03-25:
@BakaFlame I've now added the left handed / tablet invert option to the latest version [v0.1.3.](http://hwk.fi/TabletDriver/TabletDriverV0.1.3.zip)
Is that easier to understand than the tablet area rotation option?

shibou3, 2018-03-25:
@hawku Sorru for ask it in here but idk where else to ask, what is this "noise" filter?

SuperTilted, 2018-03-25:
@shibou3 It's something to remedy [the shaky cursor that happens even when the pen is stationary](https://github.com/hawku/TabletDriver/wiki/FAQ#q-why-is-the-cursor-shaking).

BakaFlame, 2018-03-26:
@hawku btw, it still doesnt match the original tablet area position! its still higher than the orignial cursor!
left handed works, but the position isnt right :((((

hawku, 2018-03-26:
@BakaFlame Oh, sorry. Didn't notice that your tablet is Wacom CTL-471

The CTL-471 tablet size configuration is wrong.
I got the settings from the Wacom Linux driver and those are wrong in there too.

Edit the tablet.cfg and change the CTL-471 configuration to this:
```
#
# Wacom CTL-471
#
Tablet 0x056a 0x0300 0xFF0D 0x01
Name "Wacom CTL-471"
ReportLength 10
ButtonMask 0x40
MaxX 15200
MaxY 9500
MaxPressure 1023
Width 152.00
Height 95.00
InitFeature 0x02 0x02
```

Those settings will also be corrected in the next driver version, which will be out in few days time.

hawku, 2018-04-01:
@BakaFlame The area size problem is fixed in the latest version.

closed 2018-04-01 by hawku