hawku/TabletDriver #75: Huion area

deymdaniel, 2018-03-26:
Will there be a huion h640p tablet area support? I'm having a hard time finding the perfect area using the wacom setting cause im using a huion tablet. Or maybe you can add huion import settings

## Hardware/software information
Windows Version: 8.1

Tablet Driver version: 

Tablet model:

deymdaniel, 2018-03-26:
Tablet model: h640p
Tablet driver version:13.15.16.180104

hawku, 2018-03-26:
I'd say that user interface of Huion's drivers is so bad that it would be pointless to try to convert the settings to this driver.

There is very high chance that the Huion's settings are already incorrect and you would just end up importing those incorrect settings.

For example when I was still using the Huion's drivers, I had to write an external application that calculates the correct aspect ratio for the Huion 420.

The drivers for H640P might be slightly better than the 420's drivers, but I really don't want to spend any more of my time on those drivers.

You should just measure the width of your H640P old area with a ruler.
Type the result of that measurement to TabletDriver's Width field and enable the force aspect ratio.