hawku/TabletDriver #79: Huion 420 not detected

Rumberkren, 2018-03-27:
i have tried:
-reinstalling vmulti driver
-reinstalling huion driver
-restarting after every installation

checked that the tablet is connected

so here's the startuplog
[startuplog.txt](https://github.com/hawku/TabletDriver/files/1850933/startuplog.txt)

## Hardware/software information
Windows Version: Windows 10 Pro 64-bit

Tablet Driver version: 0.1.3

Tablet model: Huion 420

OnoyaAn, 2018-03-27:
same for me aswell

brimstone-, 2018-03-27:
the process that worked for me with the H420 was
1. Uninstall old drivers
2. Reboot, unplug tablet before computer boots
3. Install drivers
4. Reboot
5. Plug tablet in, then start GUI

Rumberkren, 2018-03-28:
well, that's a lot of reboots again
gonna try later, thanks beforehand!

hawku, 2018-04-09:
@Rumberkren Is your tablet Huion 420 or osu!tablet?
osu!tablet's USB data is different to 420's data, so it can't be detected as Huion 420.
Check issue #99 for more information.

Rumberkren, 2018-04-14:
yeah, I have the osu!tablet. thanks for giving me the solution!

closed 2018-04-14 by Rumberkren

MarceloLuisDantas, 2021-04-28:
My tablet is Huion 420, and not work