hawku/TabletDriver #82: Screen map is not correct

Master433, 2018-03-28:
As the title says, the screen mapping isn't correct. I have set the Driver to map the whole screen but my tablet only affects the lower right portion of the screen. Not only that but I set my tablet area setting to aspect ratio but it still uses the full area.

Here's a pic of the driver settings:
[image]

## Hardware/software information
Windows Version: 10

Tablet Driver version: 0.1.3

Tablet model: XP-Pen G430S

Chrysphylax, 2018-03-28:
Did you click restart driver? You need to do that for any changes to occur

hawku, 2018-03-28:
XP-Pen G430S is not support by this driver. Currently there isn't a way to detect if the tablet is G430, G430S or G540. I will try to make a detection for the G540 when I get one in my hands.

XP-Pen is using the same USB product id for different products. That causes problems with this driver and with the XP-Pen's own drivers.

closed 2018-03-28 by hawku