hawku/TabletDriver #83: Wacom CTL480 Cursor Issue

imNP, 2018-03-29:
Hi, not sure if any one else has experienced this problem or if it's just me. Ever since I've started using this driver occasionally when I'm in the middle of a song my cursor will just randomly pull to the edge of the screen and back to where ever I'm pointing causing me to miss. It's starting to get pretty annoying and I'm not sure what is causing it. Not sure if it's just a CTL480 issue, an issue with other tablets, or just an issue on my end. My tablet settings are down below. I would try to capture video of it but I haven't had any luck yet. 

[image]

## Hardware/software information
Windows Version: Windows 10 Enterprise 64-Bit

Tablet Driver version: 0.13

Tablet model: Wacom CTL480

Kurumya, 2018-03-29:
I had the same issue several times but I think it was when my pen was too tilted.

Crymore13, 2018-03-30:
CTL480 here, using the driver since the first version and never happened here.

imNP, 2018-03-31:
I'm not tilting my pen too badly @Kurumya. How far do you you do it for it to trigger? I tried to see if I could reproduce by just using it normally but couldn't reproduce it. I can only seem to reproduce it while actually playing and holding my pen naturally so it could be that it only triggers at certain angles. I'll try to see if I can find some decent screen recording software and upload some footage when I capture it.

Dico200, 2018-03-31:
CTL480 here as well, can't say that I have this issue

hawku, 2018-03-31:
@imNP That might be a hardware defect or external signal interference.

I will try to create an optional filter that detects the USB packets with long jumps and filters those out.

imNP, 2018-03-31:
@hawku Interesting. Didn't think it could be a hardware defect or signal interference. I finally was able to capture some video of the issue and you can find it [here](https://youtu.be/-lxBoBC9pGA) (skip to about 50 seconds to see the cursor slide the right side of the screen). Also note this video was taken using driver version 0.13 so I'll go ahead and update to the latest version right not and see if the problem persists. Also note if anyone is wondering why I'm using the smoothing filter it's because for some reason my cursor shakes way too bad and I can't click anything but I assume that's normal or it's due to the signal interference.

hawku, 2018-03-31:
@imNP I watched the video and that doesn't look like a normal signal interference.

I will need more information about that problem:
- Open the Console tab and right click the console output to open a context menu.
- Select "Start Debug Log" from that menu. It will start to record debug data to a `debug_*.txt` file in the driver's folder.
- Try to reproduce the problematic pen movement while the debug log is running.
- Close the GUI when you are sure that you have captured the problem.
- Attach the `debug_*.txt` to a GitHub comment.

imNP, 2018-03-31:
@hawku Sorry for taking a little bit to get back with you. I went ahead and updated to version 0.14 and captured the debug logs like you said and have attached them here. I went ahead and got multiple of them just to be safe / thorough. Also attached my benchmark results and (what I consider) a normal debug log. Just note that when these logs occurred the cursor jumps weren't as big as what you saw in the video but the same thing happened more or less.

[Benchmark.txt](https://github.com/hawku/TabletDriver/files/1865829/Benchmark.txt)
[Debug 1.txt](https://github.com/hawku/TabletDriver/files/1865830/Debug.1.txt)
[Debug 2.txt](https://github.com/hawku/TabletDriver/files/1865831/Debug.2.txt)
[Debug 3.txt](https://github.com/hawku/TabletDriver/files/1865832/Debug.3.txt)
[Debug 4.txt](https://github.com/hawku/TabletDriver/files/1865833/Debug.4.txt)
[Debug Normal.txt](https://github.com/hawku/TabletDriver/files/1865834/Debug.Normal.txt)

imNP, 2018-03-31:
@hawku The way I ran the debug logs was I started them up and then I started playing a song on osu until  the unusual pen behavior started to occur. I was also hovering when these ran as that's how I usually play. I'll go ahead and attach some new benchmarks to make sure that the data I gave you is accurate. The pen nib is attached correctly and I'm using the pen that came with the device. If it is a hardware problem that's unfortunate for me because these tablets are a pain in the butt to find nowadays. If I remember correctly I bought this one off of eBay and it was only lightly used so who knows what the previous owner could have done to it. Not sure if it's possible to buy new ones these days.

[Benchmark.txt](https://github.com/hawku/TabletDriver/files/1865941/Benchmark.txt)

hawku, 2018-03-31:
@imNP You could try to move your tablet area to an another position.
Different parts of the tablet might have a better signal quality.

Do you have any devices close to the tablet that could interfere with the pen signal?
For example other tablets/pens, wireless chargers, electric motors, power supplies, etc.

imNP, 2018-04-01:
@hawku Not that I can think of. My tablet is pretty much in the only area it can be in really. I went ahead and moved the position of my tablet area and redid my benchmark and my numbers look similar to the ones you posted so I hope that is a good sign. I've also attached some images of my desk area (excuse the mess) as I don't see anything that could cause an interference.

https://user-images.githubusercontent.com/5455933/38168729-2758a668-351b-11e8-83ed-6a87c458d2ed.jpg)
https://user-images.githubusercontent.com/5455933/38168730-2769c588-351b-11e8-9836-9e2d82698da7.jpg

[Benchmark.txt](https://github.com/hawku/TabletDriver/files/1866023/Benchmark.txt)

Kurumya, 2018-04-03:
Did you try to change smoothing filter latency hz to differents Hz ? Because I've observed that in 125Hz, my CTL480 (S), it does the bug you described every 3 seconds.

imNP, 2018-04-04:
@Kurumya I haven't tried different smoothing filters. Since I've changed where my tablet area is it hasn't   occurred that much but I still notice it every once in a while. I just use 500 as that's what makes my cursor shake the least. It's probably an issue with my tablet but at least I know I'm not the only one experiencing it now.

imNP, 2018-04-08:
@hawku Hey just wanted to let you know I'm still having this issue. I went ahead and ordered a CTL 472 since I really couldn't find any decent 480s in my area. Once I get the 472 I'll let you know if the issue still pops up.

imNP, 2018-04-16:
@hawku Hey, just got my CTL-472 and HOLY CRAP is it a night and day difference. Before on my 480 I had to use the smooth filter just to stop it from shaking so but now I can leave it completely off and it doesn't shake nearly as bad and the cursor issue is completely gone. I just wanted to say thanks for helping me out. I wish you had a donation option because you have truly been helpful and the work you do is outstanding. If it ever happens in the future just hit me up. I'll go ahead and close this issue now. Appreciate it :).

closed 2018-04-16 by imNP

ZebastianJV, 2020-12-17:
for me its doing the same thing but with a different tablet VEIKK S640 And its doing the same thing except its not a hardware problem since i tested this with the software driver and it worked fine. but i want the hawku drivers since the software drivers that came with the tablet is so buggy and it crashes too many times