hawku/TabletDriver #88: 0xc000007b

Noaturr, 2018-03-30:
I had my system reinstalled, installed every vcredist i could find, but first i was getting an "missing vcruntime140.dll" and "missing msvcp140.dll" while trying to run drivers, so i downloaded and installed them manually, then reinstalled vmultidriver, now im getting just "application did not start correctly (0xc000007b). Press OK to end application"

## Hardware/software information
Windows Version:  10pro

Tablet Driver version: latest

Tablet model: Wacom one

Dico200, 2018-03-30:
please specify the exact model of your wacom tablet.

Noaturr, 2018-03-31:
@Dico200 I guess its CTL-471

hawku, 2018-03-31:
@Noaturr 
https://github.com/hawku/TabletDriver#installation

Noaturr, 2018-03-31:
@hawku i did it, tho it didn`t help me at all
But, like a minute ago i installed wacom drivers, then i uninstalled them, and now it works
still, thanks for the help!

closed 2018-03-31 by Noaturr