hawku/TabletDriver #90: Hope for Gaomon 1060PRO

mashirozx, 2018-03-31:
## Hardware/software information
Windows Version: Microsoft Windows NT 6.2.9200.0

Tablet Driver version: 0.1.4

Tablet model: Gaomon 1060PRO

riley-badour, 2019-01-12:
I can help you get your tablet supported i just need you to get need some information about your tablet

MyAngelXiaoYing, 2019-01-13:
> I can help you get your tablet supported i just need you to get need some information about your tablet

i can give you 1060Pro Driver and Manual about device or more i have one but i don't know how to see it.
this driver is fxxk because have two ver,one for 8192 sense and other one is nothing,i will upload two of driver.
link:https://drive.google.com/open?id=1MU-e1uv5BW0fEUkqmL48iGzdD4Up4Qu-

MyAngelXiaoYing, 2019-01-18:
> I need the startlogup.txt file. When you run the hawku tablet drivers

sure
https://drive.google.com/open?id=1LnXa4wfsqiagbDIN0G4iLyoHnmd6Qq-y

riley-badour, 2019-01-18:
Did you install the vmuti drivers also can you do the HIDlist command in the drivers and give me a copy of the output.

MyAngelXiaoYing, 2019-01-18:
> Did you install the vmuti drivers also can you do the HIDlist command in the drivers and give me a copy of the output.

wtf i install vmuti and
[2019-01-18 11:40:48.477] >> Start
[2019-01-18 11:40:48.481] TabletDriver started!
[2019-01-18 11:40:48.481] [STATUS] TABLET Unknown Huion Tablet! Check startuplog.txt!
[2019-01-18 11:40:48.481] [STATUS] USB {62F12D4C-3431-4EFD-8DD7-8E9AAB18D30C}
[2019-01-18 11:40:48.481] [STATUS] WIDTH 100.00000
[2019-01-18 11:40:48.481] [STATUS] HEIGHT 100.00000
[2019-01-18 11:40:48.481] [STATUS] MAX_X 10000
[2019-01-18 11:40:48.481] [STATUS] MAX_Y 10000
[2019-01-18 11:40:48.481] [STATUS] MAX_PRESSURE 2047
[2019-01-18 11:40:48.481] [STATUS] AUX_BUTTONS 0
[2019-01-18 11:40:48.481] >> Log Off
https://drive.google.com/open?id=18DnK3lS9-itsPQ7SGQ_UV9phiqI-Q8A0
can't move btw

MyAngelXiaoYing, 2019-01-18:
> Did you install the vmuti drivers also can you do the HIDlist command in the drivers and give me a copy of the output.

can move but just up left little area

riley-badour, 2019-01-18:
lets just say your tablet and my tablet are the same thing 1060 plus 8192
#
# Huion/Gaomon 1060PRO
#
USBTablet "{62F12D4C-3431-4EFD-8DD7-8E9AAB18D30C}"
CheckString 6 "1060PRO"
Name "Huion/Gaomon 1060PRO"
ReportId 0x08
ReportLength 8
DetectMask 0x80
IgnoreMask 0x60
MaxX 50800
MaxY 31750
MaxPressure 2047
Width 254
Height 158.75
InitStrings 100 200
AuxReportId 0x08
AuxReportLength 8
AuxDetectMask 0xE0
AuxgnoreMask 0x80
AuxCustomData ReportId Source=0
AuxCustomData ButtonsLow Source=4
AuxCustomData ButtonsHigh Source=5
AuxCustomData Detect Source=1
AuxButtonCount 12

MyAngelXiaoYing, 2019-01-18:
> lets just say your tablet and my tablet are the same thing 1060 plus 8192
>
> # Huion/Gaomon 1060PRO
> USBTablet "{62F12D4C-3431-4EFD-8DD7-8E9AAB18D30C}"
> CheckString 6 "1060PRO"
> Name "Huion/Gaomon [...]

Huion and Gaomon all just rebrand,Huion for US,Gaomon for China.
any way thank <3

riley-badour, 2019-01-18:
so wait it works how about the buttons  also does you tablet have full area 
[image]
for me my 1060 cuts out around the half was through  soft button 16
and tiny bit on the bottom is that the same for you

MyAngelXiaoYing, 2019-01-18:
> here try this one
> 
> #Huion/Gaomon 1060PRO
> 
> USBTablet "{62F12D4C-3431-4EFD-8DD7-8E9AAB18D30C}"
> CheckString 6 "1060PRO"
> Name "Huion/Gaomon 1060PRO"
> ReportId 0x07
> ReportLength 8
> DetectMask 0x80
> IgnoreMask 0x60
> MaxX 50800
> MaxY 31750
> MaxPressure 2047
> Width 254
> Height 158.75

ok this work,now i just troubled with area,gaomon driver area to wacom area.

riley-badour, 2019-01-18:
> 
> 
> so wait it works how about the buttons also does you tablet have full area
> [image]
> for me my 1060 cuts out around the half was through soft button 16
> and tiny bit on the bottom is that the same for you the blue outline

this again ^^^ also could you hit all 12 buttons with debug on without your pen  near the tablet to so i can get that working

MyAngelXiaoYing, 2019-01-19:
> > so wait it works how about the buttons also does you tablet have full area
> > [image]
> > for me my 1060 cuts out around the half was through soft button 16
> > and tiny bit on the bottom is that the same for you the blue outline
> 
> EDIT: wow so the 1060 pro is only a 8 button that's nice
> [image]
> 
> this again ^^^ also could you hit all 8 buttons with debug on without your pen near the tablet to so i can get that working but thanks so far with your data

a little problem,Gaomon 1060pro have two ver,8192 2048,my friend is 8192 ver,so....
https://drive.google.com/open?id=11ZnjRvz24mNsNnXs7Tyfgs6kyd0T98_A
his log

riley-badour, 2019-01-19:
@hawku this other person 1060 tablet is mine but its OEM02_T174 instead of Huion_174 i am thinkng they are the same tablet just with a oem identifier. Dont know if you should need another config or a check tablet string with a or Command

redolence, 2020-01-15:
Hello, by any chance you can make OEM02_T174_181125 supported ?

Stealthstarcat, 2020-02-15:
Hi,
I have a 1060 Plus. I have the button options available per using the above added aux code. I've assigned keys to them, but they don't work. What should I do different?

closed 2020-02-18 by mashirozx