hawku/TabletDriver #93: Problem with left-handed mode, wacom ctl-471

badoq, 2018-04-02:
Hey! I have annoying problem with left-handed mode. I can't use tablet on left-handed mode, even when I set 180 deg. I tried everything, any helpful advices for me please?

## Hardware/software information
Windows Version: Windows 10 NT

Tablet Driver version: 0.1.4V

Tablet model: Wacom ctl-471

hawku, 2018-04-02:
You didn't tell what is the problem?
You enabled the left handed mode and set rotation 0 and that doesn't work???

badoq, 2018-04-02:
yes

hawku, 2018-04-02:
Please, tell what is the problem.

badoq, 2018-04-02:
Tablet doesn't react for left-handed mode, for 180 deg too, it's still being right-handed.

hawku, 2018-04-02:
Did you save or apply the settings?

badoq, 2018-04-02:
Of course.

hawku, 2018-04-02:
Attach the startuplog.txt from the driver's folder to a GitHub comment

badoq, 2018-04-02:
[image]

I have something like that, but i have not any wacom driver.

badoq, 2018-04-02:
[startuplog.txt](https://github.com/hawku/TabletDriver/files/1868441/startuplog.txt)

hawku, 2018-04-02:
You have some Wacom drivers installed.
Windows might have automatically installed the Wacom drivers.
Check the issue #1

badoq, 2018-04-02:
Thanks, it works now.

closed 2018-04-02 by badoq