hawku/TabletDriver #94: "Weird area"

TensaiRhythm, 2018-04-03:
Hi there,

I have the following issue:

No matter what kind of area i pick, for example with a full tablet area, there happens to be an invisible horizontal line in the middle of the screen, top half, and bottom half of any area are completely fine. There's no issue there, in fact, i'm super happy with these drivers, but that line is a big problem, as soon as i'm at the middle of the screen, it acts like a magnet, either from middle and up or middle and down. I'd be happy to help out with some testing maybe if needed.

-IGN: Chayenne

## Hardware/software information
Windows Version: 10 Education

Tablet Driver version: 1.4 (same with 1.3)

Tablet model: Gaomon S56K

hawku, 2018-04-03:
@TensaiRhythm Does the horizontal line problem occur with other drivers?

I've heard that one of the Gaomon S56K's failure modes is that the part of the tablet area might become unusable: https://www.reddit.com/r/osugame/comments/54x267/gaomon_s56k_tablet_review_after_2_days_wacom/

TensaiRhythm, 2018-04-03:
Yes, it used to happen with the latest official drivers as well, my main concern was setting a comfortable area, but it looks like the dead zone is a straight line, maybe an actual hardware issue with these tablets?

Is there any chance that you could fix it if it wasn't an hardware issue? Thanks for your quick reaction anyways :(

hawku, 2018-04-03:
Picture of the Wacom Intuos 2 pen position detection circuit:
https://www.eevblog.com/forum/reviews/wacom-intuos-2-graphic-tablet-xd-1218-u/?action=dlattach;attach=75190;image

Gaomon S56K does have similar lines but the circuit board is flexible.
If one of those lines is broken, the tablet can't measure the pen position correctly.

Flexible circuit boards will withstand only certain amount of flexing. It might not fail immediately when bent too much, but the circuit board might harden over time and will eventually fail.

TensaiRhythm, 2018-04-03:
I never even tried to flex it, there's was and still is no point in risking anything, https://puu.sh/zVkQ1.png this is my only "workaround", simply avoid the line ;_;

Thanks again, great work on these drivers though, i tried it with my other tablets as well, but this one is just the comfiest in terms of material + response time, the combo is nice, if only it didn't have that dead zone...

closed 2018-04-09 by hawku