hawku/TabletDriver #95: Pressure Sensitivity?

syrno, 2018-04-03:
Is there a way to have pressure sensitivity and still use this driver otherwise?

## Hardware/software information
Windows Version: 10

Tablet Driver version: Newest

Tablet model: 490

hawku, 2018-04-03:
The Windows Ink mode supports pen pressure.

syrno, 2018-04-05:
Doesn't work for me.

Care to explain in more detail?

hawku, 2018-04-09:
@syrno Windows Ink is the official pen support for Windows.
https://www.microsoft.com/en-us/windows/windows-ink

If the drawing application supports Windows Ink, then the pressure sensitivity in Windows Ink mode should work with that app.

The driver doesn't support the old WinTab interface which is used by GIMP and several other digital art applications.

closed 2018-12-12 by hawku