hawku/TabletDriver #96: Just one simple ask //Linux

R1n0x, 2018-04-03:
Do you gonna do "Tablet Driver" for linux?

hawku, 2018-04-03:
Current Wacom Linux drivers can be configured so that there isn't any smoothing, so I don't see a reason to create new drivers for Linux.

closed 2018-04-03 by hawku

R1n0x, 2018-04-03:
For XP-Pen G430 ;/

StackDoubleFlow, 2020-10-11:
Wacom isn't the only tablet vendor out there, I think Linux support should at least be considered.

ArniDagur, 2020-10-11:
For what it's worth, you can use the following patch for `libinput` to disable smoothing/filtering for the likes of the XP-Pen tablet:
```diff
diff --git a/src/evdev-tablet.c b/src/evdev-tablet.c
index 67880d03..75751ee9 100644
--- a/src/evdev-tablet.c
+++ b/src/evdev-tablet.c
@@ -740,7 +740,6 @@ out:
 	}

 	tablet_history_push(tablet, &tablet->axes);
-	tablet_smoothen_axes(tablet, &axes);

 	/* The delta relies on the last *smooth* point, so we do it last */
 	axes.delta = tablet_tool_process_delta(tablet, tool, device, &axes, time);
--- a/src/filter-tablet.c
+++ b/src/filter-tablet.c
@@ -97,6 +97,7 @@ tablet_accelerator_filter_flat(struct motion_filter *filter,
 			const struct device_float_coords *units,
 			void *data, uint64_t time)
 {
+	return *(struct normalized_coords *)units;
 	struct tablet_accelerator_flat *accel_filter =
 		(struct tablet_accelerator_flat *)filter;
 	struct libinput_tablet_tool *tool = (struct libinput_tablet_tool*)data;

```