hawku/TabletDriver #98: how the fuck do tyou install

DiceSA, 2018-04-04:
none of the shit in the instlal instructions makes any fucking sense and is written in a way that implies i already know how to install it which makes installation instructions fucking pointl,ess. the fuck is , "you need to run install_huion_64.bat, which is in the driver_huion directory."

the fuck is install_huion_64.bat 

what is that?

where is that?

I dont understand. why cant you just make a fucking program that works. jesus chbrist

LeoFLT, 2018-04-04:
Wow there buddy, seems like you're a little triggered. First off, make sure you download the compiled link (http://hwk.fi/TabletDriver/TabletDriverV0.1.4.zip), otherwise there's not gonna be such a folder, the rest is just as it's written, not hard at all.

imNP, 2018-04-04:
@DiceSA You just need to follow the [installation instructions](https://github.com/hawku/TabletDriver/blob/master/README.md#installation) which aren't even that hard (extract the folder anywhere, run the install.bat file, launch the tablet gui exe). If you can't do that then you need to take a computer 101 class or something because this program is not that hard to install.

DiceSA, 2018-04-04:
https://imgur.com/a/WeXt8

tell me where install is and you win a cash prize