hawku/TabletDriver #99: is official osu!tablet supported?

Syetriz, 2018-04-04:
is official osu!tablet supported? since it reskin of huion 420?

## Hardware/software information
Windows Version: 8.1

Tablet Driver version: v0.1.4

Tablet model: official osu!tablet (huion 420?)

SuperTilted, 2018-04-04:
Yes, it should. [The Huion 420 is under supported tablets.](https://github.com/hawku/TabletDriver#supported-tablets)

> Supported tablets:
    Wacom CTL-470
    Wacom CTL-471
    Wacom CTL-472
    Wacom CTL-480
    Wacom CTH-480
    Wacom CTL-490
    XP Pen G430 (New 2017+ "Model B")
    XP Pen G640
    **Huion 420**
    Huion H640P
    Gaomon S56K

Syetriz, 2018-04-05:
@hawku
Thanks it started working with "Tablet "{62F12D4C-3431-4EFD-8DD7-8E9AAB18D30C}" 200 HVAN"

closed 2018-04-05 by Syetriz