hawku/TabletDriver #100: Strange cursor movement

Banbeucmas, 2018-04-05:
Similar to https://github.com/hawku/TabletDriver/issues/83
But with mouse 
I cannot provide the log tho. But I reckon that is because the smoothing filter would continue running even if the tablet ism't being detected.

## Hardware/software information
Windows Version: Window 8.1 64bit

Tablet Driver version: 0.1.4

Tablet model: Mouse, No tablet was plugged

hawku, 2018-04-06:
@Banbeucmas The smoothing filter can't do anything to the mouse input.
When the tablet isn't detected the TabletDriver will do a tablet device search every ~5 seconds.
Does the mouse problem occur every 5 seconds?

Banbeucmas, 2018-04-07:
Around that. But only osu! interface have this issue

closed 2019-01-09 by hawku