hawku/TabletDriver #101: High Liftoff Causing Cursor Teleport

HyperSlayer72, 2018-04-05:
Video: https://www.youtube.com/watch?v=dt_bpJCNPVQ
This happens regardless of where I put my tablet reigon, doesn't happen on latest wacom driver.
This does happen during gameplay causing some annoying misses.

## Hardware/software information
Windows Version: Windows 10 Professional 64-bit

Tablet Driver version: 0.1.4

Tablet model: CTH-490

hawku, 2018-04-05:
Create a debug log:
https://github.com/hawku/TabletDriver/issues/83#issuecomment-377709586

HyperSlayer72, 2018-04-05:
[debug_2018-04-05_01_18_57.txt](https://github.com/hawku/TabletDriver/files/1881042/debug_2018-04-05_01_18_57.txt)
Here you go:

HyperSlayer72, 2018-04-05:
Also Just a heads up my pen is permanently clicked.

hawku, 2018-04-05:
@HyperSlayer72 Did you try the Wacom driver with permanently clicked tip?

Taping the tip might lower the maximum hover height.

HyperSlayer72, 2018-04-05:
Welp, your right, it still happens with the wacom driver. Any Way of fixing this? considering id still like the hardware level smoothing to be disabled.

HyperSlayer72, 2018-04-05:
Another thing, for the ctl 490 on your tablet test webpage could you show tests for taped down nib while hovering? Just want to see it from a test perspective.

hawku, 2018-04-06:
@HyperSlayer72 The "drag" test result is done with ~1 mm hover and a taped tip:
http://hwk.fi/TabletTest/wacom490drag_12000.png

I think that the best way to fix the Wacom 490's problems is to buy a Wacom CTL-472...
Gameplay-wise the CTL-472 is basically the same as CTL-490 but doesn't have the same hardware problems or smoothing.

closed 2018-04-06 by hawku