hawku/TabletDriver #103: Smoothing permanently on in Windows 7

mosvannen, 2018-04-07:
I tried using the driver on both my Windows 10 laptop my Windows 7 desktop, but on Windows 7 the driver seems to always smooth out inputs. On Windows 10 it behaves as it should, like the official 6.3.9w5 driver (without prediction). 

I've checked the console and it says `Smoothing = off`, and turning the smoothing filter on via the checkbox also doesn't make it feel any different than having it off.

I've also tested this with my CTL-471, with the same behavior discrepancies on the different OSes.

## Hardware/software information
Windows Version: Windows 7 x64

Tablet Driver version: 0.1.4

Tablet model: Wacom CTH-480/CTL-471

mosvannen, 2018-04-08:
I'm using 0 ms latency, as I want it to be as responsive as possible. When I put the latency to 100ms it works as expected (makes the cursor extremely slow and smooth).

What I'm trying to say is that this:
[image]
leads to the cursor motion still feeling slightly smoothed out instead of "raw input".

hawku, 2018-04-09:
@mosvannen Does that happen with the Wacom 6.3.9w5 drivers?
Did you disable the Windows Aero?

mosvannen, 2018-04-09:
The Wacom 6.3.9w5 drivers make the cursor very responsive and "raw" feeling, but with the cost of prediction. The 0.1.4 tablet driver in comparison just feels much more sluggish on Windows 7 compared to Windows 10. In addition, the Wacom drivers behave and feel the exact same on both OSes. 

I also have tried using the Classic, Basic, and Aero themes to no avail (DWM on/off changes nothing).

mosvannen, 2018-04-10:
I don't know if it's placebo but the Windows Ink mode seems to increase the smoothing, but Relative doesn't feel any different other than behaving as a mouse. Setting smoothing to 1ms@125Hz doesn't change anything really.

Sorry for not being able to describe the issue better, but I'm sure that there's a difference to how the driver works/feels on Windows 7 vs Windows 10 (somehow).

mosvannen, 2018-04-13:
Okay, so I discovered that in order to achieve the "raw" feeling of the tablet you have to first install and then uninstall Wacom drivers. Installing this tablet driver on a fresh Windows installation leads to the cursor movement feeling weird, as I described.

If you can't fix this, then it would be nice if you put some sort of disclaimer on the readme.

closed 2018-04-13 by mosvannen