hawku/TabletDriver #106: Cursor is fading movement when pen is stopped

IrtenaX, 2018-04-08:
## Hardware/software information
Windows Version: 10 64bit Home

Tablet Driver version: 0.14

Tablet model: huion h640p
[Untitled Project.zip](https://github.com/hawku/TabletDriver/files/1887202/Untitled.Project.zip)

hawku, 2018-04-08:
@IrtenaX Does that happen with the official drivers?

IrtenaX, 2018-04-09:
Yes

hawku, 2018-04-09:
@IrtenaX It has to be a hardware issue if the tablet doesn't work correctly even with the Huion's official drivers.

IrtenaX, 2018-04-09:
Can you do something with this because it's some smoothering code or should I write to the huion support?

hawku, 2018-04-09:
Yes, you should send a message to Huion support.

Huion H640P and H430P does have the highest amount of hardware input smoothing of all tablets that I have tested.
I've contacted Huion about that, but haven't got any reply yet.

IrtenaX, 2018-04-09:
Ok thank you for the interest

closed 2018-08-12 by hawku