hawku/TabletDriver #107: Save multiple profiles?

Frubbs, 2018-04-09:
Perhaps it would be beneficial to incorporate multiple profiles that you can save either to a specific application or just manually hot-swap in the app each time you want to change it. Either way would be much faster than what I currently having to do which is:

Change my Wacom area dimensions, change mode, change button mapping, and change smoothing filter, takes about a minute or two each time!

Thanks if you are willing to incorporate the idea!

## Hardware/software information
Windows Version: 

Tablet Driver version: 

Tablet model:

hawku, 2018-04-09:
@Frubbs You can export the settings to a file. Main menu -> File -> Export

I'm planning to add profiles and it will need large changes to the GUI code, so I might have to rewrite most of it.

Emi-Chan, 2018-04-11:
I can't seem to find any "Main Menu", so.. am i just too stupid? >.<

hawku, 2018-04-12:
@Emi-Chan By main menu I meant the menu bar above the Settings tab.
```
[TD] TabletDriverGUI - Wacom XYZ-123
File <---- That one!
----------.
 Settings |  Commands | Console 
          '----------------------
```

Emi-Chan, 2018-04-12:
Oh. My bad. I was running 0.1.0 for some reason. So the File Menu wasn't there. My bad. Thanks anyway!

closed 2018-08-12 by hawku