hawku/TabletDriver #112: Tablet not found (Huion 420 Bought from the osu store)

LukeBrandenburg, 2018-04-10:
## Hardware/software information
Windows Version: 10 64bit

Tablet Driver version: 0.1.4

Tablet model:  H420

Keep getting the error that the tablet isnt found.

2018-04-10 18:44:50 [INFO]   Driver version: 0.1.4
2018-04-10 18:44:50 [INFO]   Windows version: Microsoft Windows NT 6.2.9200.0
2018-04-10 18:44:50 [INFO]   Windows product: Windows 10 Education
2018-04-10 18:44:50 [INFO]   Windows release: 1709
2018-04-10 18:44:50 [INFO] 
2018-04-10 18:44:50 [INFO] >> CheckTablet
2018-04-10 18:44:50 [ERROR] Tablet not found!
2018-04-10 18:44:50 [ERROR] Check the list of supported tablets from the GitHub page.
2018-04-10 18:44:50 [ERROR] http://github.com/hawku/TabletDriver

I have included the startup log under here.
[startuplog.txt](https://github.com/hawku/TabletDriver/files/1895546/startuplog.txt)

hawku, 2018-04-10:
Try this:
https://github.com/hawku/TabletDriver/issues/99#issuecomment-378952466

LukeBrandenburg, 2018-04-10:
That solved it. Thank you!

closed 2018-04-10 by LukeBrandenburg