hawku/TabletDriver #113: Driver found tablet but settings don't apply to the tablet

Desaje13, 2018-04-11:
## Hardware/software information
Windows Version: 10 64bit

Tablet Driver version: 0.1.4

Tablet model: Wacom CTL-472

imNP, 2018-04-11:
@Desaje13 You'll need to provide some more information. Did you you hit the "Apply" and "Save" buttons after changing the settings you needed to? I know for certain settings unless you do that they won't take effect unless you do that or until you hit the "Restart Driver" button.

Desaje13, 2018-04-11:
@imNP It tried to "Apply","Save","Restart" moreover I reinstalled driver about 4-5 times.  Problems appeared after 2 days of using.

hawku, 2018-04-11:
@Desaje13 Does it say that "Wacom drivers installed" on the window title?
If it does then check the issue #1

closed 2018-08-12 by hawku