hawku/TabletDriver #118: Tablet Not found (CTL-4100)

Noerusagi, 2018-04-14:
## Hardware/software information
Windows Version: Windows 10 1709 x64

Tablet Driver version: 0.1.4

Tablet model: CTL-4100

I included the startuplog in here
[startuplog.txt](https://github.com/hawku/TabletDriver/files/1909977/startuplog.txt)

Noerusagi, 2018-04-14:
hmmm, It doesn't work...

2018-04-14 20:15:40 [INFO] >> #
2018-04-14 20:15:40 [INFO] >> # Wacom CTL-4100
2018-04-14 20:15:40 [INFO] >> #
2018-04-14 20:15:40 [INFO] >> Tablet 0x056a 0x0376 0xFF0D 0x0001
2018-04-14 20:15:40 [WARNING] Can't open HID tablet 0x056A 0x0376 0xFF0D 0x0001
2018-04-14 20:15:40 [INFO] >> Tablet 0x056a 0x0374 0xFF0D 0x0001
2018-04-14 20:15:40 [WARNING] Can't open HID tablet 0x056A 0x0374 0xFF0D 0x0001
2018-04-14 20:15:40 [INFO] >> Name "Wacom CTL-4100"
2018-04-14 20:15:40 [INFO] >> ReportId 0x10
2018-04-14 20:15:40 [INFO] >> ReportLength 192
2018-04-14 20:15:40 [INFO] >> DetectMask 0x40
2018-04-14 20:15:40 [INFO] >> MaxX 15200
2018-04-14 20:15:40 [INFO] >> MaxY 9500
2018-04-14 20:15:40 [INFO] >> MaxPressure 4095
2018-04-14 20:15:40 [INFO] >> Width 152.0
2018-04-14 20:15:40 [INFO] >> Height 95.0
2018-04-14 20:15:40 [INFO] >> Type Wacom4100
2018-04-14 20:15:40 [INFO] >> #
2018-04-14 20:15:40 [INFO] >> # Wacom CTL-4100 (Wacom drivers installed)
2018-04-14 20:15:40 [INFO] >> #
2018-04-14 20:15:40 [INFO] >> Tablet 0x056a 0x0376 0xFF0D 0x0001
2018-04-14 20:15:40 [WARNING] Can't open HID tablet 0x056A 0x0376 0xFF0D 0x0001
2018-04-14 20:15:40 [INFO] >> Tablet 0x056a 0x0374 0xFF0D 0x0001
2018-04-14 20:15:40 [WARNING] Can't open HID tablet 0x056A 0x0374 0xFF0D 0x0001
2018-04-14 20:15:40 [INFO] >> Name "Wacom CTL-4100 (Wacom drivers installed)"
2018-04-14 20:15:40 [INFO] >> ReportId 0x10
2018-04-14 20:15:40 [INFO] >> ReportLength 193
2018-04-14 20:15:40 [INFO] >> DetectMask 0x40
2018-04-14 20:15:40 [INFO] >> MaxX 15200
2018-04-14 20:15:40 [INFO] >> MaxY 9500
2018-04-14 20:15:40 [INFO] >> MaxPressure 4095
2018-04-14 20:15:40 [INFO] >> Width 152.0
2018-04-14 20:15:40 [INFO] >> Height 95.0
2018-04-14 20:15:40 [INFO] >> Type Wacom4100
2018-04-14 20:15:40 [INFO] >> #
2018-04-14 20:15:40 [INFO] >> # Wacom CTL-4100 Bluetooth
2018-04-14 20:15:40 [INFO] >> Tablet 0x056a 0x0377 0xFF0D 0x0001
2018-04-14 20:15:40 [WARNING] Can't open HID tablet 0x056A 0x0377 0xFF0D 0x0001
2018-04-14 20:15:40 [INFO] >> Name "Wacom CTL-4100 Bluetooth (Experimental)"
2018-04-14 20:15:40 [INFO] >> ReportId100 Bluetooth
2018-04-14 20:15:40 [WARNING] Unknown command: ReportId100 Bluetooth
2018-04-14 20:15:40 [INFO] >> # 0x81
2018-04-14 20:15:40 [INFO] >> ReportLength 361
2018-04-14 20:15:40 [INFO] >> DetectMask 0xE0
2018-04-14 20:15:40 [INFO] >> MaxX 15200
2018-04-14 20:15:40 [INFO] >> MaxY 9500
2018-04-14 20:15:40 [INFO] >> MaxPressure 4095
2018-04-14 20:15:40 [INFO] >> Width 152.0
2018-04-14 20:15:40 [INFO] >> Height 95.0

There is a little problem in the network, so it can't be upload startuplog.txt

Noerusagi, 2018-04-14:
Hardware ID is USB\VID_056A&PID_0374&REV_0106

Noerusagi, 2018-04-14:
hmm..... this problem solved, and i finding it is a minor mistake I made.
I found that I had installed the wacom driver before.
So it's not {
Tablet 0x056a 0x0376 0xFF0D 0x0001
Tablet 0x056a 0x0374 0xFF0D 0x0001
}
It's {
Tablet 0x056a 0x0376 0xFF00 0x000A
Tablet 0x056a 0x0374 0xFF00 0x000A
}
That's the right thing to do

hawku, 2018-04-14:
Uninstall or disable the Wacom drivers. This driver doesn't work when the Wacom drivers are running.

If you want to try to run this driver while the Wacom drivers are installed, run the `tools\disable_wacom_drivers.bat` and edit the tablet.cfg to look like this:
```
#
# Wacom CTL-4100
#
Tablet 0x056a 0x0376 0xFF0D 0x0001
Tablet 0x056a 0x0374 0xFF0D 0x0001
Name "Wacom CTL-4100"
ReportId 0x10
ReportLength 192
DetectMask 0x40
MaxX 15200
MaxY 9500
MaxPressure 4095
Width 152.0
Height 95.0
Type Wacom4100

#
# Wacom CTL-4100 (Wacom drivers installed)
#
Tablet 0x056a 0x0376 0xFF00 0x000A
Tablet 0x056a 0x0374 0xFF00 0x000A
Name "Wacom CTL-4100 (Wacom drivers installed)"
ReportId 0x10
ReportLength 193
DetectMask 0x40
MaxX 15200
MaxY 9500
MaxPressure 4095
Width 152.0
Height 95.0
Type Wacom4100
```

The best is just to uninstall the Wacom drivers. Running two drivers at the same time will cause some problems.

Noerusagi, 2018-04-14:
ok All problem solved, Thanks

closed 2018-04-14 by hawku